mikalhart/TinyGPSPlus #10: 'radians' and 'degrees' function not declared

creativeknights2012, 2015-04-28:
I'm getting compiling errors when I try to compile Arduino: 1.6.3 (Windows 7), Board: "Intel® Edison"

TinyGPS++.cpp: In static member function 'static double TinyGPSPlus::distanceBetween(double, double, double, double)':

TinyGPS++.cpp:290:39: error: 'radians' was not declared in this scope

   double delta = radians(long1 - long2);

                                       ^

TinyGPS++.cpp: In static member function 'static double TinyGPSPlus::courseTo(double, double, double, double)':

TinyGPS++.cpp:314:38: error: 'radians' was not declared in this scope

   double dlon = radians(long2 - long1);

                                      ^

TinyGPS++.cpp:325:20: error: 'degrees' was not declared in this scope

   return degrees(a2);

                    ^

Error compiling.

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.
 this library using Arduino 1.6.3 on an Intel Edison board.

creativeknights2012, 2015-05-15:
This issue isn't just constrained for the Intel Edison board either. The arduino library has completely removed the radian and degree math functions.

creativeknights2012, 2015-05-15:
Problem solved. Just needed to change them to cosine math functions.

closed 2015-05-15 by creativeknights2012

dokterbob, 2016-03-21:
@creativeknights2012 What exactly did you change here? I'm having the same problem and I fail to understand how TinyGPSPlus is apparently still using these removed Arduino functions.

jorgemzz, 2016-09-24:
I am porting TinyGPSPlus for linux, I've already port some arduino specific functions like millis() for linux, now I have that radians() was not declared. I've just realize its a function to convert lat(DDDD.DD degree + decimal degree format) to radians. To solve that issue would be as easy to define a function to convert degrees to radians.