mikalhart/TinyGPSPlus #12: gps.location.isValid() method returns true when fix is lost

sycophantic, 2015-12-19:
I'm not sure if this a feature or a bug.  It appears that gps.location.isValid() returns true even when the fix is lost for some time.  Does it not reset if a fix is lost?

Using `if (gps.location.isValid() &&  gps.location.age() < 2000)` is my work around.

stilldavid, 2016-02-14:
This is a feature - the library doesn't know if the location that the GPS is reporting is valid or not beyond the checksum passing.

Your workaround is the correct way of handling outdated (and potentially wrong, thought not "invalid") fixes.

mikalhart, 2017-04-14:
@stilldavid is correct.  isValid() simply means that the location information contains valid (though possibly stale) data.  isValid() returns FALSE until the very first NMEA sentence with a fix is parsed.  From then on it returns TRUE forever.

closed 2017-04-14 by mikalhart