mikalhart/TinyGPSPlus #13: How can I save the obtained location & then send it through SMS?

Pritha01, 2016-04-16:
Hello! I have following function:

`bool sendLocation(char* number, char* latitude, char* longtitude)
{
    cleanBuffer();
    serialSIM800.println("AT+CMGF=1");
    if ( waitFor("OK", "ERROR") != 1 ) return false;

    cleanBuffer();
    serialSIM800.print("AT+CMGS=\"");
    serialSIM800.print(number);
    serialSIM800.println("\"");
    if ( waitFor(">", "ERROR") != 1 ) return false;

    cleanBuffer();
    serialSIM800.print("LAT: ");
    serialSIM800.println(latitude);
    serialSIM800.print("LON: ");
    serialSIM800.println(longtitude);

    serialSIM800.println((char)26);
    if ( waitFor("+CMGS:", "ERROR") != 1 ) return false;

    return true;
}
`

Will it be fine if I define latitude & longitude as:
char* latitude;
char* longtitude;

How can I save the obtained latitude & longitude and then pass them into my function? Please help me.

nobodyguy, 2017-01-07:
If you're using AVR, you can do something like this:
```c++
char latBuffer[20];
char lngBuffer[20];
double lat = gps.location.lat();
double lng = gps.location.lng();
dtostrf(lat, 8, 6, latBuffer);
dtostrf(lng, 8, 6, lngBuffer);
```

gamecompiler, 2017-03-14:
Thank you very much for this code nobodyguy, id got frustraded using sprintf......
Funny fact is that id not searched for this in context with sms.
I want to send gps coordinates over LoRa.

nobodyguy, 2017-03-15:
You can check code for my LoRaWAN GPS tracker prototype if you want - https://gist.github.com/nobodyguy/7920482dd00f300b973126f9d7188db4

closed 2017-04-14 by mikalhart