mikalhart/TinyGPSPlus #16: GNNS instead GPS

mclSaxony, 2016-07-06:
for some years I heared the term Global Navigation Satellite System (GNNS) and found this as an naming-superset of Global Positioning Systems. It made no difference for me because I used only US GPS satellites.

Now the times have gone foreward and other GNNS like Glonass in Europe also play an important role in navigation.

With modern uBlox receivers also one problem arises:
  ! The NMEA sentences are not longer named GPxyz if other GNNS are included in the solution !

This means one can rely only on ariving GPGGA sentences if ONLY GPS is enabled. Doing this would be foolish in Europe because we will loose about one half of the satellites in view.

So I changed the processing of the NMEA sentences. Because the GNNS specific sentences only differ in the 2nd char ( 'P'= U.S.-GPS, 'L'=Glonass, 'B'= Baidou, 'N' = generic ) I replace the 2nd char of the NMEA sentence with 'N' as the generic char and compare the sentences with "GNxyz". So the detection still works. Alls variables I adopted as well  for this GNNS naming scheme...

 Pitfall: ublox has introduced/reserved NMEA sentences with only one underscore instead the "GN" ( or "GP" ) prefix. Such messages I have never seen in reality but they would break my solution ... but the build in parser as well.

[TinyGPSPlus.zip](https://github.com/mikalhart/TinyGPSPlus/files/349506/TinyGPSPlus.zip)

q2dg, 2016-07-27:
Mikal, are you still here?

nseidle, 2017-04-13:
I too had this issue. More and more modules output GNGGA and GNRMC.

@q2dg - your solution would cause the library to not work with GPS that output GPGGA sentences. My recommendation:

    #define _GPRMCterm   "GPRMC"
    #define _GNRMCterm   "GNRMC"
    #define _GPGGAterm   "GPGGA"
    #define _GNGGAterm   "GNGGA"

with a few other bits. Pull request is here: https://github.com/mikalhart/TinyGPSPlus/pull/28

mikalhart, 2017-04-14:
Nate's pull request was merged.

closed 2017-04-14 by mikalhart