mikalhart/TinyGPSPlus #17: Support for AI-Thinker A7 modue

gorlapraveen, 2016-08-12:


nasimulhaque, 2017-01-21:
I brought an A7 module but didn't find any code to use it. Is there any library for that anywhere?

danieldbb, 2017-01-23:
Hi,
does anyone have SDK for A7?

I notice several problems with different A7.
i have 12 modules, from two diff suppliers. Only one module doesn't accept AT commands. Reset is ok, I can trace traffic but when I try simple "AT\n\r" nothing hapends ... as port were dead, burned out (URTX). I used Logic analyzer to check the signal from connector to the chip .. null, nil, nada ... like there is none.
Beside those A7 I have they react differently at Reset ... pull down circuit with N-MOS Fet ... sometimes POWER LED turns OFF, sometimes not. Sometimes A7 booted, sometimes not.
Another problem I have is:

- Phone book Write Command: AT+CPBW=index,"<number>",129,"name"
if there is no index field supplied by write command A7 doesn't accept this entry and reboots

- Phone book Find Command: AT+CPBF="search string" A7 performs searching only bye name not also by number

- AT+CLIP=1 displays only the caller number and not also his name from phone book (???)

I don't know if I have the last Revision? Support from A7 doesn't responds to my emails.
Therefore I would also like to have A7 SDK!
With failures like this, we should have a A7 Errors forum.

knowing how to avoid those failures, working with A7 is good. But not perfect :(

What is your experience with A7?

Regards,
Daniel - See more at: http://www.esp8266.com/viewtopic.php?f=159&t=13411&p=61400#p61400

mikalhart, 2017-03-25:
You should be able to use TinyGPS++.  The NMEA stream is the same as always.
HOWEVER!  I have been playing unsuccessfully with the A7.  I've got three: one "dead" as you describe, and the other two reboot every time I send an SMS from my cell phone to the module!

mikalhart, 2017-04-14:
Verified: TinyGPS++ works with the A7.

closed 2017-04-14 by mikalhart

nasimulhaque, 2017-04-15:
Thank you :-) Is there any tutorial available for that?

gamecompiler, 2017-06-30:
Easy to use.
Just send 
```

AT+GPS=1
AT+AGPS=1
AT+GPSRD=1
```

But does anyone know how to let the a7 put the NMEA sentences to the other uart port? Got some problems when using TinyGPS while gps is enabled

adamjade, 2017-12-13:
Hi, the string received starts with "GPRSD " word which is not expected by the read method