mikalhart/TinyGPSPlus #19: Calculate heading to coordinate

gamecompiler, 2017-01-02:
Hi, id ran into the problem that im not able to calculate the heading from a coordinate to a given coordinate (fastest way, air routing)if im "on the other side of the world". Ive got a HMC5883 compass from which im able to calculate the heading in degrees to North. 

At this time im calculating the heading using Pythagoras formula.
How would you calculate the heading if you are "on the other side of the world" ?
Thats a very interesting math question ^^ I thought about "splitting" the triangle into two parts to the "other side of the world".

Im sorry for my english. (Do you speak german?).
Thanks in advance.

mikalhart, 2017-01-03:
Hi!  Are you using the TinyGPS++ "courseTo" function?  Is it not working?

gamecompiler, 2017-01-03:
Yes i tryed it, but it worked not very good, or i didnt understood it.
My gps is currently in germany - huchting.
[image]

The heading of 275 degrees is correct, but what would be if im on the left side of the world and my target on the right, the fastest route would be to go left, but my code goes to the right (longest route...)

https://upload.wikimedia.org/wikipedia/commons/0/09/BlankMap-World-v2.png

Who do you calculate the course in this case?

computergeek125, 2017-02-19:
Something I have been experimenting with is normalizing the GPS heading from a 0-360 range to a -180 to +180 range, where -90 represents west (or left) at 270 degrees, +90 represents east (or right) at 90 degrees, and ±180 represents south.  That way, you can use a negative value to represent left.  Haven't proven that the algorithm we're using works 100% of the time, but as soon as we do (or at least most of the time), I'll post it here.

gamecompiler, 2017-02-19:
The best way should be to put "the same world" left, right, up and down. Calculate the heading to those worlds and take a look which direction is the shortest.
If id understood your idea correctly, the issue to calculate the shortest path isnt solved.

For example when youre at greenwhich, and you want to go to china, there are two ways to go to china.
But the problem id got is to calculate those ways and try which is the fastest.

The main problem id got is math - my math isnt good enough to calculate the fastest way using euler formula (just straigth to china istn the fastest way and not the correct path because our world isnt flat). That theme is very interessting for me because im trying to build my own geocaching gps.

mikalhart, 2017-10-05:
This is an interesting conversation, but I don't think there is a TinyGPS++ issue, so I'm going to close it.

closed 2017-10-05 by mikalhart

computergeek125, 2017-10-13:
Probably a fair call.  For now, it looks like more of a mathematics problem rather than a code issue.  I guess we could readdress the issue once the math is solved?