mikalhart/TinyGPSPlus #26: Number of Satellites and altitude

Hachmamut, 2017-03-31:
I'm trying to get the number of used satellites and the altitude from my GPS modul, a ublox NEO 6M.

Reading the blank Serial data, I can see that both values a in the GPGGA Sentence.
But using the functions of your library results in "0" in both cases. Any idea what the problem is?
Best regards

Felix

mikalhart, 2017-04-14:
Can you please post the GPGGA sentence and the code you are using (that returns the "0")?

AEK-BKF, 2017-09-28:
Hi,

Any news about your issue ? I have the same mistake.

Thanks.

AEK-BKF, 2017-09-28:
Never mind, it works :) just wait for few moments :)

closed 2017-10-05 by mikalhart

Metaln00b, 2022-04-10:
It is still an issue.

Using `TinyGPSCustom customAltitude(gps, "GPGGA", 9);` and `TinyGPSCustom customSattelites(gps, "GPGGA", 7);` the value is empty, too. GPPGA shows both values with fix flag = 1.

Any solution?

Metaln00b, 2022-04-13:
Setting the Baud to 9600 and set "protocol in" from "0+1+5 - UBX+NMEA+RTCM3" to "0+1 - UBX+NMEA" via u-center solves the problem.