mikalhart/TinyGPSPlus #30: GPS Time 3 seconds fast!?

thorntonforce, 2017-05-22:
Firstly thankyou very much for this excellent library, it has proved incredibly useful for my latest project. This is more of a question than an issue as I don't think my problem is directly related to any of the API's in this library....but its very confusing and I didnt know where else to post the question.

I have made a race timing system consisting of two identical boxes - one is a start gate and one is a finish gate. To sync the time on both gates I am using Neo-6m GPS modules like this.... [http://www.ebay.co.uk/itm/172406984538?_trksid=p2057872.m2749.l2649&ssPageName=STRK%3AMEBIDX%3AIT](url)
connected to a an Arduino Pro Micro.

On startup I wait for a valid and updated time string and then read the hour(), minute(), second() and then start an on board timer to keep time. I dont look for any more GPS messages after this point but I do wait for the PPS interrupt. Once I have this I update the millisecond count for better accuracy and then periodically check this interrupt to keep everything nicely synced.

It all works great and I used the system successfully at the weekend for a friendly bike race. However I have one annoying issue that I cant work out. Both boxes are identical in terms of hardware and software (although I got the 2 Neo-6m boards from different ebay sellers and they look slightly different.) But they behave in a slightly different way on startup. If I am outside with a good view of the sky both boxes acquire the time after a few seconds and its accurate (within a second) to the correct GPS time taken from the internet. After a minute or so we see the PPS signal and the times are synced exactly.

All well and good. However...occasionally outside but always inside (e.g when the GPS signal is weaker) - if I power on the boxes one of them works ok but takes a little longer to connect - but the other one (and its always the same one) starts ticking 3 seconds fast.

I have checked the data I am reading from the TinyGPSPlus API's and this is where the 3 second difference comes from.

Eventually if I take the box outside I start reading the correct time but since I only read the time once this is a problem for me as I have to keep power cycling it until it reads the correct time.

Its always the same box and always 3 seconds which makes it very confusing and also because the time is 3 seconds FAST. If it was slow I would assume I was reading old data.

Its almost as if the satellite itself is fast - which is of course impossible. 

So I am pretty stumped by this one - I just thought I would post it here as you of course have a lot of experience with reading the GPS messages - have you ever seen this before?

Thanks
Nick

bal4, 2017-06-02:
Currently got the same issue. Neo 6m runs 3 seconds fast with the library on an esp8266 and ublox 7 is bang on on a pro micro. Intermittent problem. I'm looking into it.

thorntonforce, 2017-06-05:
Great - please let me know if you have success - Im still stumped!

thorntonforce, 2017-06-05:
I just discovered this....

https://forum.u-blox.com/index.php?qa=5504&qa_1=neo-module-rom-03-gpgga-time-data-is-always-faster-seconds-why

and this...

https://forum.u-blox.com/index.php?qa=6319&qa_1=neo-m8n-1s-time-shift-after-fix-for-5m

Interesting !
Haven't had time to look for a solution yet - but this is definitely the problem!

bal4, 2017-06-06:
I'm on the latest firmware (Version: 7.03 (45970) Mar 17 2011 16:26:24), and have the issue.  This is a firmware issue not dealing with leap second not a tinygps issue.  The forum also explains that after 12.5 min the gps should correct for the leap second but that is too long for me.

mikalhart, 2017-10-05:
Interesting!  Closing for now.

closed 2017-10-05 by mikalhart

cadr, 2024-11-29:
Hi.  That forum doesn't seem to be around anymore - do you happen to know the issue/resolution?