mikalhart/TinyGPSPlus #34: Zip File of Release 0.95 contains Version 0.92

newbee42, 2017-07-17:
When I download the zip files V0.95 to install, then i can see in TinyGPS++.h the Version 0.92 and in cpp i can not see any GN Support. A weak work around is to copy alle the Files manually from https://github.com/mikalhart/TinyGPSPlus/

Newbee42

jvhaarst, 2017-09-27:
I just checked, the release zip file which is marked 0.95 does contain the latest code, only the version number in TinyGPS++.h is wrong.
If you want the release with the right version in the TinyGPS++.h file, you can download the 0.95a release, as that does contain the right file.

It's a bit confusion, just use 0.95a, that one is the same as the current master.

mikalhart, 2017-10-05:
Ugh.  Sorry.  Fixed in 1.0.0 (just posted).

closed 2017-10-05 by mikalhart