mikalhart/TinyGPSPlus #37: Why the glonass support was removed?

tuloski, 2017-12-21:
As the title...

mikalhart, 2017-12-22:
Hmm..  Seems to have been in error.  Fixed in 1.0.1.  Thanks for the note.

closed 2017-12-22 by mikalhart

FLYINGSC0T, 2019-05-21:
Hi,

Can anyone help ?
Have upgraded to v 1.0.2 but running SatelliteTracker.ino does not pick up GLONASS sentences. Had the same problem with v 0.92.

GPS module is UBlox M8N and output from UBlox Center (GPS configured to receive GPS, GLONASS and GALILEO with NMEA version 4.01) is:

Output from UBlox software GPS=$GPGSV, GLONASS=$GLGSV, GALILEO=$GAGSV:
06:07:22  $GPGSV,4,1,14,05,07,187,,10,07,329,,11,00,038,,12,09,212,23,0*66
06:07:22  $GPGSV,4,2,14,13,66,128,26,15,74,273,16,17,27,105,35,19,18,129,08,0*6F
06:07:22  $GPGSV,4,3,14,20,21,306,,24,40,278,19,28,39,051,24,30,06,080,35,0*6D
06:07:22  $GPGSV,4,4,14,36,32,146,,49,38,181,,0*69
06:07:22  $GLGSV,3,1,11,65,59,298,17,66,05,333,,71,11,166,18,72,56,193,,0*74
06:07:22  $GLGSV,3,2,11,73,20,051,30,74,62,021,24,75,48,259,25,76,07,244,,0*78
06:07:22  $GLGSV,3,3,11,82,04,360,,83,15,045,,84,06,098,30,0*44
06:07:22  $GAGSV,3,1,10,01,68,235,26,04,62,284,29,09,15,317,,13,00,088,,0*77
06:07:22  $GAGSV,3,2,10,14,19,069,,15,04,041,,19,46,165,14,21,55,051,26,0*7D
06:07:22  $GAGSV,3,3,10,27,04,053,25,31,17,233,,0*73

...clearly showing that the 3 satellite systems are being received.

The output from SatelliteTracker.ino is:

Output from SatelliteTracker.ino with "GPGSV":
Sats=12 Nums=10 12 13 15 17 19 20 24 25 28 32 36  Elevation=19 47 25 50 35 42 16 74 10 6 0 32  Azimuth=296 231 150 183 61 90 268 319 235 51 321 146  SNR=0 24 24 32 27 30 0 24 0 16 0 0 
Sats=12 Nums=10 12 13 15 17 19 20 24 25 28 32 36  Elevation=19 47 25 50 35 42 16 74 10 6 0 32  Azimuth=296 231 150 183 61 90 268 319 235 51 321 146  SNR=0 25 25 31 28 30 0 24 0 15 0 0 
Sats=12 Nums=10 12 13 15 17 19 20 24 25 28 32 36  Elevation=19 47 25 50 35 42 16 74 10 6 0 32  Azimuth=296 231 150 183 61 90 268 319 235 51 321 146  SNR=0 25 25 31 27 29 0 23 0 14 0 0 
Sats=12 Nums=10 12 13 15 17 19 20 24 25 28 32 36  Elevation=19 47 25 50 35 42 16 74 10 6 0 32  Azimuth=296 231 150 183 61 90 268 319 235 51 321 146  SNR=0 25 25 31 27 29 0 23 0 14 0 0 
Sats=12 Nums=10 12 13 15 17 19 20 24 25 28 32 36  Elevation=19 47 25 50 35 42 16 74 10 6 0 32  Azimuth=296 231 150 183 61 90 268 319 235 51 321 146  SNR=0 26 25 31 27 29 0 23 0 15 0 0 
Sats=12 Nums=10 12 13 15 17 19 20 24 25 28 32 36  Elevation=19 47 25 50 35 42 16 74 10 6 0 32  Azimuth=296 231 150 183 61 90 268 319 235 51 321 146  SNR=0 26 25 31 28 29 0 23 0 15 0 0 

Output from SatelliteTracker.ino with "GLGSV":

Sats=7 Nums= Elevation= Azimuth= SNR=
Sats=10 Nums= Elevation= Azimuth= SNR=
Sats=12 Nums= Elevation= Azimuth= SNR=
Sats=12 Nums= Elevation= Azimuth= SNR=
Sats=12 Nums= Elevation= Azimuth= SNR=
Sats=12 Nums= Elevation= Azimuth= SNR=
Sats=12 Nums= Elevation= Azimuth= SNR=
Sats=12 Nums= Elevation= Azimuth= SNR=
Sats=12 Nums= Elevation= Azimuth= SNR=

Output from SatelliteTracker.ino with "GAGSV":
Sats=12 Nums=1 4 5 9 13 15 19 21 26 31  Elevation=75 50 0 39 18 1 10 24 16 44  Azimuth=26 219 330 299 59 14 171 65 110 263  SNR=11 19 0 0 19 0 0 0 21 13 
Sats=12 Nums=1 4 5 9 13 15 19 21 26 31  Elevation=75 50 0 39 18 1 10 24 16 44  Azimuth=26 219 330 299 59 14 171 65 110 263  SNR=10 18 0 0 19 0 0 0 20 13 
Sats=12 Nums=1 4 5 9 13 15 19 21 26 31  Elevation=75 50 0 39 18 1 10 24 16 44  Azimuth=26 219 330 299 59 14 171 65 110 263  SNR=10 19 0 0 19 0 0 0 20 11 
Sats=12 Nums=1 4 5 9 13 15 19 21 26 31  Elevation=75 50 0 39 18 1 10 24 16 44  Azimuth=26 219 330 299 59 14 171 65 110 263  SNR=15 19 0 0 18 0 0 0 20 13 
Sats=12 Nums=1 4 5 9 13 15 19 21 26 31  Elevation=75 50 0 39 18 1 10 24 16 44  Azimuth=26 219 330 299 59 14 171 65 110 263  SNR=19 18 0 0 19 0 0 0 19 14 
Sats=12 Nums=1 4 5 9 13 15 19 21 26 31  Elevation=75 50 0 39 18 1 10 24 16 44  Azimuth=26 219 330 299 59 14 171 65 110 263  SNR=16 18 0 0 18 0 0 0 20 14 
Sats=12 Nums=1 4 5 9 13 15 19 21 26 31  Elevation=75 50 0 39 18 1 10 24 16 44  Azimuth=26 219 330 299 59 14 171 65 110 263  SNR=7 17 0 0 18 0 0 0 21 15 

...No data for Nums, Elevation,  Azimuth or SNR !!

The strings  "GPGSV",  "GLGSV", "GAGSV" were changed accordingly in the sketch.
Also, in the UBlox center more than 20 satellites are found but in TINYGPS++ it is much less.

Thanks for any help on this and would be appreciated.