mikalhart/TinyGPSPlus #45: GPS data is not changing

pesengineeringclub, 2018-05-16:
Hi 
First of all, thanks for the library. when we test it, the GPS data is not changing even we moving far away. Do you have any idea about the issue? thanks.

we use  Seeeduino LoRaWAN W/GPS
https://www.seeedstudio.com/Seeeduino-LoRaWAN-W%2FGPS-p-2781.html

mikalhart, 2019-10-13:
Can you examine the raw NMEA code?  This library merely parses the raw NMEA.

aliexclusive, 2019-11-04:
Hi

I got nice raw NMEA codes. 
[image]

But when I test DeviceExample, it returns the following error.

Location: INVALID  Date/Time: INVALID INVALID

mikalhart, 2019-11-05:
Hi Muhammed—

 

If you’re printing the raw NMEA while processing them, you may have created a delay that causes overflow.  Also, check to make sure baud and port are correct in the program.

 

 <https://user-images.githubusercontent.com/23704914/68143411-71949c80-ff42-11e9-8668-260bf79d97b5.jpeg>

closed 2021-07-13 by mikalhart