mikalhart/TinyGPSPlus #48: custom line GPGST works but when it is GNGST it does not

KommandorKeen, 2018-08-09:
I update my GPS firmware and now it spits out GNGST rather than GPGST.
When it was GPGST the following worked:
TinyGPSCustom posrms(gps, "GPGST", 2);
TinyGPSCustom latsig(gps, "GPGST", 6);
TinyGPSCustom lonsig(gps, "GPGST", 7);
TinyGPSCustom htsig(gps, "GPGST", 8);

but as 
TinyGPSCustom posrms(gps, "GNGST", 2);
TinyGPSCustom latsig(gps, "GNGST", 6);
TinyGPSCustom lonsig(gps, "GNGST", 7);
TinyGPSCustom htsig(gps, "GNGST", 8);

The GNSS unit is outputting 
$GNGST,093415.00,2.963,,,,0.466,0.421,0.819*60

Eheran1, 2018-12-31:
This is working for me:
TinyGPSCustom errorlat(gps, "GNGST", 6);    //=1 sigma error
TinyGPSCustom errorlong(gps, "GNGST", 7);   //=1 sigma error
TinyGPSCustom errorheight(gps, "GNGST", 8); //=1 sigma error

Example sentence:
$GNGST,114341.00,2827,,,,1.8,1.2,1.7*48
errorlat = 1.8
errorlong = 1.2
errorheight = 1.7

Eheran1, 2018-12-31:
Same here:
TinyGPSCustom errorlat(gps, "GNGST", 6);    //=1 sigma error
TinyGPSCustom errorlong(gps, "GNGST", 7);   //=1 sigma error
TinyGPSCustom errorheight(gps, "GNGST", 8); //=1 sigma error

With sentence:
$GNGST,114341.00,2827,,,,1.8,1.2,1.7*48

Results in:
errorlat.value() = 2.7
errorlong.value() = 2.1
errorheight.value() = 2.7
These numbers never change. Its always 2.7, 2.1 and 2.7 instead of eg. 1.8, 1.2 and 1.7.

mikalhart, 2019-01-02:
Interesting.  I'd like to do some experimenting with this.  Could one of you possibly provide a stream of sample sentences--maybe 15 or 20--that I could run through my test engine?  Thanks.

Eheran1, 2019-01-02:
Due to Checksum I didnt alter the position information. Please delete this after reading it or let me know so I can do that.

`$GNRMC,094507.00,A,5121.23052,N,01159.45724,E,0.030,,020119,,,D,V*1B
$GNVTG,,T,,M,0.030,N,0.055,K,D*3B
$GNGNS,094507.00,5121.23052,N,01159.45724,E,DAAN,21,0.81,94.3,45.0,,0000,V*2F
$GNGGA,094507.00,5121.23052,N,01159.45724,E,2,12,0.81,94.3,M,45.0,M,,0000*76
$GNGSA,A,3,07,03,23,26,16,22,09,06,19,,,,1.33,0.81,1.06,1*0F
$GNGSA,A,3,76,65,67,75,81,77,,,,,,,1.33,0.81,1.06,2*05
$GNGSA,A,3,05,03,08,24,25,31,,,,,,,1.33,0.81,1.06,3*01
$GNGLL,5121.23052,N,01159.45724,E,094507.00,A,D*70
$GNGST,094507.00,34,,,,1.3,1.5,2.5*40
$GNZDA,094507.00,02,01,2019,00,00*7E
$GNRMC,094508.00,A,5121.23063,N,01159.45710,E,0.065,,020119,,,D,V*11
$GNVTG,,T,,M,0.065,N,0.120,K,D*38
$GNGNS,094508.00,5121.23063,N,01159.45710,E,DAAN,21,0.74,94.1,45.0,,0000,V*2D
$GNGGA,094508.00,5121.23063,N,01159.45710,E,2,12,0.74,94.1,M,45.0,M,,0000*74
$GNGSA,A,3,07,03,23,26,16,22,09,06,19,,,,1.27,0.74,1.02,1*04
$GNGSA,A,3,76,65,67,75,81,77,,,,,,,1.27,0.74,1.02,2*0E
$GNGSA,A,3,05,03,08,24,25,31,,,,,,,1.27,0.74,1.02,3*0A
$GNGLL,5121.23063,N,01159.45710,E,094508.00,A,D*7A
$GNGST,094508.00,31,,,,1.3,1.5,2.5*4A
$GNZDA,094508.00,02,01,2019,00,00*71
$GNRMC,094509.00,A,5121.23071,N,01159.45700,E,0.097,,020119,,,D,V*1F
$GNVTG,,T,,M,0.097,N,0.179,K,D*39
$GNGNS,094509.00,5121.23071,N,01159.45700,E,DAAN,21,0.74,93.8,45.0,,0000,V*20
$GNGGA,094509.00,5121.23071,N,01159.45700,E,2,12,0.74,93.8,M,45.0,M,,0000*79
$GNGSA,A,3,07,03,23,26,16,22,09,06,19,,,,1.27,0.74,1.02,1*04
$GNGSA,A,3,76,65,67,75,81,77,,,,,,,1.27,0.74,1.02,2*0E
$GNGSA,A,3,05,03,08,24,25,31,,,,,,,1.27,0.74,1.02,3*0A
$GNGLL,5121.23071,N,01159.45700,E,094509.00,A,D*79
$GNGST,094509.00,27,,,,1.3,1.4,2.4*4C
$GNZDA,094509.00,02,01,2019,00,00*70
$GNRMC,094510.00,A,5121.23076,N,01159.45698,E,0.151,,020119,,,D,V*1B
$GNVTG,,T,,M,0.151,N,0.280,K,D*37
$GNGNS,094510.00,5121.23076,N,01159.45698,E,DAAN,22,0.74,93.7,45.0,,0000,V*23
$GNGGA,094510.00,5121.23076,N,01159.45698,E,2,12,0.74,93.7,M,45.0,M,,0000*79
$GNGSA,A,3,07,03,23,26,16,22,09,06,19,,,,1.27,0.74,1.02,1*04
$GNGSA,A,3,76,65,66,67,75,81,77,,,,,,1.27,0.74,1.02,2*0E
$GNGSA,A,3,05,03,08,24,25,31,,,,,,,1.27,0.74,1.02,3*0A
$GNGLL,5121.23076,N,01159.45698,E,094510.00,A,D*76
$GNGST,094510.00,26,,,,1.3,1.4,2.4*45
$GNZDA,094510.00,02,01,2019,00,00*78
$GNRMC,094511.00,A,5121.23083,N,01159.45684,E,0.081,,020119,,,D,V*11
$GNVTG,,T,,M,0.081,N,0.151,K,D*34
$GNGNS,094511.00,5121.23083,N,01159.45684,E,DAAN,23,0.76,93.5,45.0,,0000,V*24
$GNGGA,094511.00,5121.23083,N,01159.45684,E,2,12,0.76,93.5,M,45.0,M,,0000*7F
$GNGSA,A,3,07,03,23,26,16,22,09,06,19,,,,1.36,0.76,1.13,1*06
$GNGSA,A,3,76,65,66,82,67,75,81,77,,,,,1.36,0.76,1.13,2*06
$GNGSA,A,3,05,03,08,24,25,31,,,,,,,1.36,0.76,1.13,3*08
$GNGLL,5121.23083,N,01159.45684,E,094511.00,A,D*70
$GNGST,094511.00,25,,,,1.3,1.4,2.4*47
$GNZDA,094511.00,02,01,2019,00,00*79
$GNRMC,094512.00,A,5121.23085,N,01159.45681,E,0.105,,020119,,,D,V*1C
$GNVTG,,T,,M,0.105,N,0.194,K,D*30
$GNGNS,094512.00,5121.23085,N,01159.45681,E,DAAN,23,0.68,93.3,45.0,,0000,V*2D
$GNGGA,094512.00,5121.23085,N,01159.45681,E,2,12,0.68,93.3,M,45.0,M,,0000*76
$GNGSA,A,3,07,03,23,26,16,22,09,06,19,,,,1.11,0.68,0.87,1*00
$GNGSA,A,3,76,65,66,82,67,75,81,77,,,,,1.11,0.68,0.87,2*00
$GNGSA,A,3,05,03,08,24,25,31,,,,,,,1.11,0.68,0.87,3*0E
$GNGLL,5121.23085,N,01159.45681,E,094512.00,A,D*70
$GNGST,094512.00,31,,,,1.3,1.4,2.4*41
$GNZDA,094512.00,02,01,2019,00,00*7A
$GNRMC,094513.00,A,5121.23082,N,01159.45693,E,0.183,,020119,,,D,V*17
$GNVTG,,T,,M,0.183,N,0.340,K,D*35
$GNGNS,094513.00,5121.23082,N,01159.45693,E,DAAN,23,0.77,93.4,45.0,,0000,V*21
$GNGGA,094513.00,5121.23082,N,01159.45693,E,2,12,0.77,93.4,M,45.0,M,,0000*7A
$GNGSA,A,3,07,03,23,26,16,22,09,06,19,,,,1.21,0.77,0.93,1*08
$GNGSA,A,3,76,65,66,82,67,75,81,77,,,,,1.21,0.77,0.93,2*08
$GNGSA,A,3,05,03,08,24,25,31,,,,,,,1.21,0.77,0.93,3*06
$GNGLL,5121.23082,N,01159.45693,E,094513.00,A,D*75
[... 790 lines cut ...]
$GAGSV,3,3,09,31,06,018,21,0*43
$GNGLL,5121.23044,N,01159.45856,E,094534.00,A,D*7D
$GNGST,094534.00,21,,,,1.2,1.3,2.1*47
$GNZDA,094534.00,02,01,2019,00,00*7E
$GNRMC,094535.00,A,5121.23039,N,01159.45867,E,0.012,,020119,,,D,V*1F
$GNVTG,,T,,M,0.012,N,0.022,K,D*3B
$GNGNS,094535.00,5121.23039,N,01159.45867,E,DAAN,23,0.80,85.4,45.0,,0000,V*2F
$GNGGA,094535.00,5121.23039,N,01159.45867,E,2,12,0.80,85.4,M,45.0,M,,0000*74
$GNGSA,A,3,07,03,23,26,16,22,09,06,19,,,,1.27,0.80,0.99,1*0C
$GNGSA,A,3,76,65,66,82,67,75,81,77,,,,,1.27,0.80,0.99,2*0C
$GNGSA,A,3,05,03,08,24,25,31,,,,,,,1.27,0.80,0.99,3*02
$GPGSV,4,1,15,02,32,307,,03,26,116,40,06,54,257,31,07,32,177,43,0*63
$GPGSV,4,2,15,09,83,248,35,16,11,080,40,19,05,239,20,22,07,119,40,0*6B
$GPGSV,4,3,15,23,63,070,38,26,14,053,30,29,02,355,,30,04,195,37,0*6E
$GPGSV,4,4,15,33,26,214,,36,28,156,48,39,,,40,0*6D
$GLGSV,3,1,09,65,10,089,33,66,65,051,26,67,47,295,22,75,33,044,28,0*76
$GLGSV,3,2,09,76,44,103,29,77,10,160,33,81,14,217,35,82,38,267,22,0*73
$GLGSV,3,3,09,83,27,327,,0*48
$GAGSV,3,1,09,03,86,021,32,05,33,082,41,08,36,268,26,13,36,294,16,0*70
$GAGSV,3,2,09,15,34,229,40,24,29,065,28,25,25,122,43,26,06,337,,0*75
$GAGSV,3,3,09,31,06,018,20,0*42
$GNGLL,5121.23039,N,01159.45867,E,094535.00,A,D*74
$GNGST,094535.00,23,,,,1.2,1.3,2.1*44
$GNZDA,094535.00,02,01,2019,00,00*7F
`

mikalhart, 2019-01-02:
Thanks.  I removed the location data.

mikalhart, 2019-01-02:
I'm starting to wonder whether the problems you guys are experiencing are due to dropping characters.

I made a test program that statically parses the GPS data and it seems to work fine.

Could you monitor the gps.failedChecksum() count?  That number should be 0 or at least very small and non-increasing over time in a properly working sketch.

My test program, using the data @Eheran1 posted earlier:
```
#include <TinyGPS++.h>
TinyGPSPlus gps;

char data[] = 
"$GNRMC,094507.00,A,XXXX.XXXXX,N,XXXXX.XXXXX,E,0.030,,020119,,,D,V*1B\r\n"
"$GNVTG,,T,,M,0.030,N,0.055,K,D*3B\r\n"
<big snip>
"$GNGST,094535.00,23,,,,1.2,1.3,2.1*44\r\n"
"$GNZDA,094535.00,02,01,2019,00,00*7F\r\n";

TinyGPSCustom errorlat(gps, "GNGST", 6); //=1 sigma error
TinyGPSCustom errorlong(gps, "GNGST", 7); //=1 sigma error
TinyGPSCustom errorheight(gps, "GNGST", 8); //=1 sigma error

void setup() 
{
  delay(5000);
  Serial.begin(115200);
}

const char *p = data;

void loop() 
{
  if (*p)
    gps.encode(*p++);
  
  // Every time anything is updated, print it.
  if (gps.altitude.isUpdated())
  {
    Serial.print(F("ALT=")); Serial.print(gps.altitude.meters()); 
  }
  
  if (gps.satellites.isUpdated())
  {
    Serial.print(F(" SATS=")); Serial.println(gps.satellites.value());
  }
  
  if (errorlat.isUpdated())
  {
     Serial.print(F(" ERRLT=")); Serial.println(errorlat.value());
  }

  if (errorlong.isUpdated())
  {
     Serial.print(F(" ERRLG=")); Serial.println(errorlong.value());
  }
  
  if (errorheight.isUpdated())
  {
     Serial.print(F(" ERRHT=")); Serial.println(errorheight.value());
  }
}
```

This generates the following output:

ALT=94.30 SATS=12
 ERRLT=1.3
 ERRLG=1.5
 ERRHT=2.5
ALT=94.10 SATS=12
 ERRLT=1.3
 ERRLG=1.5
 ERRHT=2.5
ALT=93.80 SATS=12
 ERRLT=1.3
 ERRLG=1.4
 ERRHT=2.4
ALT=93.70 SATS=12
 ERRLT=1.3
 ERRLG=1.4
 ERRHT=2.4
ALT=93.50 SATS=12
 ERRLT=1.3
 ERRLG=1.4
 ERRHT=2.4
ALT=93.30 SATS=12
 ERRLT=1.3
 ERRLG=1.4
 ERRHT=2.4
ALT=93.40 SATS=12
 ERRLT=1.3
 ERRLG=1.4
 ERRHT=2.3
ALT=93.40 SATS=12
 ERRLT=1.2
 ERRLG=1.3
 ERRHT=2.3
ALT=93.20 SATS=12
 ERRLT=1.2
 ERRLG=1.3
 ERRHT=2.3
ALT=93.10 SATS=12
 ERRLT=1.2
 ERRLG=1.3
 ERRHT=2.3
ALT=92.80 SATS=12
 ERRLT=1.2
 ERRLG=1.3
 ERRHT=2.3
ALT=92.40 SATS=12
 ERRLT=1.2
 ERRLG=1.3
 ERRHT=2.3
ALT=92.20 SATS=12
 ERRLT=1.2
 ERRLG=1.3
 ERRHT=2.3
ALT=91.90 SATS=12
 ERRLT=1.2
 ERRLG=1.3
 ERRHT=2.3
ALT=91.70 SATS=12
 ERRLT=1.2
 ERRLG=1.3
 ERRHT=2.2
ALT=91.30 SATS=12
 ERRLT=1.2
 ERRLG=1.3
 ERRHT=2.2
ALT=90.90 SATS=12
 ERRLT=1.2
 ERRLG=1.3
 ERRHT=2.2
ALT=90.60 SATS=12
 ERRLT=1.2
 ERRLG=1.3
 ERRHT=2.2
ALT=90.30 SATS=12
 ERRLT=1.2
 ERRLG=1.3
 ERRHT=2.2
ALT=89.70 SATS=12
 ERRLT=1.2
 ERRLG=1.3
 ERRHT=2.2
ALT=89.30 SATS=12
 ERRLT=1.2
 ERRLG=1.3
 ERRHT=2.2
ALT=89.00 SATS=12
 ERRLT=1.2
 ERRLG=1.3
 ERRHT=2.2
ALT=88.70 SATS=12
 ERRLT=1.2
 ERRLG=1.3
 ERRHT=2.2
ALT=88.20 SATS=12
 ERRLT=1.2
 ERRLG=1.3
 ERRHT=2.2
ALT=87.70 SATS=12
 ERRLT=1.2
 ERRLG=1.3
 ERRHT=2.1
ALT=86.80 SATS=12
 ERRLT=1.2
 ERRLG=1.3
 ERRHT=2.1
ALT=86.30 SATS=12
 ERRLT=1.2
 ERRLG=1.3
 ERRHT=2.1
ALT=85.60 SATS=12
 ERRLT=1.2
 ERRLG=1.3
 ERRHT=2.1
ALT=85.40 SATS=12
 ERRLT=1.2
 ERRLG=1.3
 ERRHT=2.1

mikalhart, 2019-01-02:
I count 584 lines.  Here's mine:

```
"$GNRMC,094507.00,A,5121.23052,N,01159.45724,E,0.030,,020119,,,D,V*1B\r\n"
"$GNVTG,,T,,M,0.030,N,0.055,K,D*3B\r\n"
"$GNGNS,094507.00,5121.23052,N,01159.45724,E,DAAN,21,0.81,94.3,45.0,,0000,V*2F\r\n"
"$GNGGA,094507.00,5121.23052,N,01159.45724,E,2,12,0.81,94.3,M,45.0,M,,0000*76\r\n"
"$GNGSA,A,3,07,03,23,26,16,22,09,06,19,,,,1.33,0.81,1.06,1*0F\r\n"
"$GNGSA,A,3,76,65,67,75,81,77,,,,,,,1.33,0.81,1.06,2*05\r\n"
"$GNGSA,A,3,05,03,08,24,25,31,,,,,,,1.33,0.81,1.06,3*01\r\n"
"$GPGSV,4,1,15,02,32,307,,03,26,116,47,06,55,258,34,07,31,177,48,0*67\r\n"
"$GPGSV,4,2,15,09,83,247,28,16,11,080,41,19,06,240,23,22,08,119,43,0*6B\r\n"
"$GPGSV,4,3,15,23,64,070,34,26,14,053,33,29,01,356,,30,03,195,35,0*63\r\n"
"$GPGSV,4,4,15,33,26,214,,36,28,156,49,39,,,41,0*6D\r\n"
"$GLGSV,3,1,09,65,10,089,33,66,65,051,,67,47,295,25,75,33,044,21,0*7C\r\n"
"$GLGSV,3,2,09,76,44,103,39,77,10,160,30,81,14,217,36,82,38,267,,0*72\r\n"
"$GLGSV,3,3,09,83,27,327,22,0*48\r\n"
"$GAGSV,3,1,09,03,86,021,29,05,33,082,42,08,36,268,29,13,36,294,27,0*74\r\n"
"$GAGSV,3,2,09,15,34,229,38,24,29,065,32,25,25,122,44,26,06,337,,0*76\r\n"
"$GAGSV,3,3,09,31,06,018,22,0*40\r\n"
"$GNGLL,5121.23052,N,01159.45724,E,094507.00,A,D*70\r\n"
"$GNGST,094507.00,34,,,,1.3,1.5,2.5*40\r\n"
"$GNZDA,094507.00,02,01,2019,00,00*7E\r\n"
"$GNRMC,094508.00,A,5121.23063,N,01159.45710,E,0.065,,020119,,,D,V*11\r\n"
"$GNVTG,,T,,M,0.065,N,0.120,K,D*38\r\n"
"$GNGNS,094508.00,5121.23063,N,01159.45710,E,DAAN,21,0.74,94.1,45.0,,0000,V*2D\r\n"
"$GNGGA,094508.00,5121.23063,N,01159.45710,E,2,12,0.74,94.1,M,45.0,M,,0000*74\r\n"
"$GNGSA,A,3,07,03,23,26,16,22,09,06,19,,,,1.27,0.74,1.02,1*04\r\n"
"$GNGSA,A,3,76,65,67,75,81,77,,,,,,,1.27,0.74,1.02,2*0E\r\n"
"$GNGSA,A,3,05,03,08,24,25,31,,,,,,,1.27,0.74,1.02,3*0A\r\n"
"$GPGSV,4,1,15,02,32,307,,03,26,116,45,06,54,257,32,07,31,177,48,0*6D\r\n"
"$GPGSV,4,2,15,09,83,247,23,16,11,080,41,19,06,240,16,22,08,119,41,0*64\r\n"
"$GPGSV,4,3,15,23,64,070,35,26,14,053,33,29,01,356,,30,03,195,35,0*62\r\n"
"$GPGSV,4,4,15,33,26,214,,36,28,156,49,39,,,40,0*6C\r\n"
"$GLGSV,3,1,09,65,10,089,31,66,65,051,26,67,47,295,21,75,33,044,29,0*76\r\n"
"$GLGSV,3,2,09,76,44,103,39,77,10,160,28,81,14,217,35,82,38,267,23,0*79\r\n"
"$GLGSV,3,3,09,83,27,327,17,0*4E\r\n"
"$GAGSV,3,1,09,03,86,021,27,05,33,082,40,08,36,268,27,13,36,294,23,0*72\r\n"
"$GAGSV,3,2,09,15,34,229,37,24,29,065,32,25,25,122,43,26,06,337,,0*7E\r\n"
"$GAGSV,3,3,09,31,06,018,18,0*49\r\n"
"$GNGLL,5121.23063,N,01159.45710,E,094508.00,A,D*7A\r\n"
"$GNGST,094508.00,31,,,,1.3,1.5,2.5*4A\r\n"
"$GNZDA,094508.00,02,01,2019,00,00*71\r\n"
"$GNRMC,094509.00,A,5121.23071,N,01159.45700,E,0.097,,020119,,,D,V*1F\r\n"
"$GNVTG,,T,,M,0.097,N,0.179,K,D*39\r\n"
"$GNGNS,094509.00,5121.23071,N,01159.45700,E,DAAN,21,0.74,93.8,45.0,,0000,V*20\r\n"
"$GNGGA,094509.00,5121.23071,N,01159.45700,E,2,12,0.74,93.8,M,45.0,M,,0000*79\r\n"
"$GNGSA,A,3,07,03,23,26,16,22,09,06,19,,,,1.27,0.74,1.02,1*04\r\n"
"$GNGSA,A,3,76,65,67,75,81,77,,,,,,,1.27,0.74,1.02,2*0E\r\n"
"$GNGSA,A,3,05,03,08,24,25,31,,,,,,,1.27,0.74,1.02,3*0A\r\n"
"$GPGSV,4,1,15,02,32,307,,03,26,116,45,06,54,257,32,07,32,177,47,0*61\r\n"
"$GPGSV,4,2,15,09,83,247,29,16,11,080,40,19,06,240,13,22,08,119,40,0*6B\r\n"
"$GPGSV,4,3,15,23,64,070,35,26,14,053,33,29,01,356,,30,03,195,36,0*61\r\n"
"$GPGSV,4,4,15,33,26,214,,36,28,156,48,39,,,39,0*63\r\n"
"$GLGSV,3,1,09,65,10,089,30,66,65,051,24,67,47,295,21,75,33,044,30,0*7D\r\n"
"$GLGSV,3,2,09,76,44,103,37,77,10,160,29,81,14,217,35,82,38,267,23,0*76\r\n"
"$GLGSV,3,3,09,83,27,327,20,0*4A\r\n"
"$GAGSV,3,1,09,03,86,021,29,05,33,082,41,08,36,268,27,13,36,294,24,0*7A\r\n"
"$GAGSV,3,2,09,15,34,229,37,24,29,065,27,25,25,122,43,26,06,337,,0*7A\r\n"
"$GAGSV,3,3,09,31,06,018,17,0*46\r\n"
"$GNGLL,5121.23071,N,01159.45700,E,094509.00,A,D*79\r\n"
"$GNGST,094509.00,27,,,,1.3,1.4,2.4*4C\r\n"
"$GNZDA,094509.00,02,01,2019,00,00*70\r\n"
"$GNRMC,094510.00,A,5121.23076,N,01159.45698,E,0.151,,020119,,,D,V*1B\r\n"
"$GNVTG,,T,,M,0.151,N,0.280,K,D*37\r\n"
"$GNGNS,094510.00,5121.23076,N,01159.45698,E,DAAN,22,0.74,93.7,45.0,,0000,V*23\r\n"
"$GNGGA,094510.00,5121.23076,N,01159.45698,E,2,12,0.74,93.7,M,45.0,M,,0000*79\r\n"
"$GNGSA,A,3,07,03,23,26,16,22,09,06,19,,,,1.27,0.74,1.02,1*04\r\n"
"$GNGSA,A,3,76,65,66,67,75,81,77,,,,,,1.27,0.74,1.02,2*0E\r\n"
"$GNGSA,A,3,05,03,08,24,25,31,,,,,,,1.27,0.74,1.02,3*0A\r\n"
[... 491 lines cut ...]
"$GNGST,094534.00,21,,,,1.2,1.3,2.1*47\r\n"
"$GNZDA,094534.00,02,01,2019,00,00*7E\r\n"
"$GNRMC,094535.00,A,5121.23039,N,01159.45867,E,0.012,,020119,,,D,V*1F\r\n"
"$GNVTG,,T,,M,0.012,N,0.022,K,D*3B\r\n"
"$GNGNS,094535.00,5121.23039,N,01159.45867,E,DAAN,23,0.80,85.4,45.0,,0000,V*2F\r\n"
"$GNGGA,094535.00,5121.23039,N,01159.45867,E,2,12,0.80,85.4,M,45.0,M,,0000*74\r\n"
"$GNGSA,A,3,07,03,23,26,16,22,09,06,19,,,,1.27,0.80,0.99,1*0C\r\n"
"$GNGSA,A,3,76,65,66,82,67,75,81,77,,,,,1.27,0.80,0.99,2*0C\r\n"
"$GNGSA,A,3,05,03,08,24,25,31,,,,,,,1.27,0.80,0.99,3*02\r\n"
"$GPGSV,4,1,15,02,32,307,,03,26,116,40,06,54,257,31,07,32,177,43,0*63\r\n"
"$GPGSV,4,2,15,09,83,248,35,16,11,080,40,19,05,239,20,22,07,119,40,0*6B\r\n"
"$GPGSV,4,3,15,23,63,070,38,26,14,053,30,29,02,355,,30,04,195,37,0*6E\r\n"
"$GPGSV,4,4,15,33,26,214,,36,28,156,48,39,,,40,0*6D\r\n"
"$GLGSV,3,1,09,65,10,089,33,66,65,051,26,67,47,295,22,75,33,044,28,0*76\r\n"
"$GLGSV,3,2,09,76,44,103,29,77,10,160,33,81,14,217,35,82,38,267,22,0*73\r\n"
"$GLGSV,3,3,09,83,27,327,,0*48\r\n"
"$GAGSV,3,1,09,03,86,021,32,05,33,082,41,08,36,268,26,13,36,294,16,0*70\r\n"
"$GAGSV,3,2,09,15,34,229,40,24,29,065,28,25,25,122,43,26,06,337,,0*75\r\n"
"$GAGSV,3,3,09,31,06,018,20,0*42\r\n"
"$GNGLL,5121.23039,N,01159.45867,E,094535.00,A,D*74\r\n"
"$GNGST,094535.00,23,,,,1.2,1.3,2.1*44\r\n"
"$GNZDA,094535.00,02,01,2019,00,00*7F\r\n";
```

Interesting that your failedChecksum counter increases during the run with the STATIC data above.  That is definitely weird.  What microcontroller are you using?  (I.e. you need more than just Uno to store all this data, which overflows the RAM.)

Eheran1, 2019-01-03:
I deleted all the (here useless) *GSV data to save RAM. Im using a adafruit feather m0.
I c&p the data now, had to crop a little to make it fit in RAM and it did work!
I gathered some new NMEA data and put it in there, worked too!

Then I changed something: I added "TinyGPSCustom satnr(gps, "GNGNS", 7);" and corresponding "if (satnr.isUpdated())  {    Serial.print(F("  realsat=")); Serial.print(satnr.value());  }"
Still works.

THEN I put all the prints into one "if", and that was "if (gps.speed.isUpdated())"
This broke it. This, however, workes, only changed "speed" to "altitude":
```  
if (gps.altitude.isUpdated())
  {    
    Serial.print(F("ALT="));        Serial.print(gps.altitude.meters());
    Serial.print(F("  SATS="));     Serial.print(gps.satellites.value());
    Serial.print(F("  realsat="));  Serial.print(satnr.value());
    Serial.print(F("  ERRLT="));    Serial.print(errorlat.value());
    Serial.print(F("  ERRLG="));    Serial.print(errorlong.value());
    Serial.print(F("  ERRHT="));    Serial.println(errorheight.value());  }
```
Its also broken with "gps.location.isUpdated()" and "gps.time.isUpdated()" and gps.course.isUpdated() and gps.hdop.isUpdated(). 
It works with said "gps.altitude.isUpdated()" and "gps.satellites.isUpdated()".

Since I need to do everything with every updated GPS-point and the main object is speed... this is a issue for me now. Altitude might stay the same while speed and position changes. Only time, position and speed are values that have to define a "new position" to display/log etc. - idealy its time, since that always iterates up with a new GPS-solution/position.

This is how it looks like when its working (first values missing since they come after the altitude-value, the others are the previous ones):
```
5s
Start
ALT=110.10  SATS=12  realsat=14  ERRLT=  ERRLG=  ERRHT=
ALT=110.00  SATS=12  realsat=15  ERRLT=3.5  ERRLG=2.8  ERRHT=4.9
ALT=109.60  SATS=12  realsat=15  ERRLT=2.9  ERRLG=2.4  ERRHT=4.2
ALT=109.20  SATS=12  realsat=15  ERRLT=2.6  ERRLG=2.2  ERRHT=3.9
ALT=108.50  SATS=12  realsat=14  ERRLT=2.6  ERRLG=2.2  ERRHT=3.8
ALT=108.10  SATS=12  realsat=14  ERRLT=2.4  ERRLG=2.1  ERRHT=3.5
ALT=108.10  SATS=12  realsat=14  ERRLT=2.2  ERRLG=2.0  ERRHT=3.3
ALT=107.90  SATS=11  realsat=11  ERRLT=2.2  ERRLG=2.0  ERRHT=3.3
ALT=107.80  SATS=10  realsat=10  ERRLT=2.1  ERRLG=2.0  ERRHT=3.3
ALT=106.70  SATS=12  realsat=15  ERRLT=2.1  ERRLG=1.9  ERRHT=3.3
ALT=106.70  SATS=12  realsat=15  ERRLT=1.9  ERRLG=1.9  ERRHT=3.1
ALT=106.60  SATS=12  realsat=16  ERRLT=1.9  ERRLG=1.8  ERRHT=3.1
ALT=101.80  SATS=12  realsat=19  ERRLT=1.9  ERRLG=1.8  ERRHT=3.1
ALT=101.60  SATS=12  realsat=19  ERRLT=1.6  ERRLG=1.4  ERRHT=2.6
ALT=101.40  SATS=12  realsat=19  ERRLT=1.5  ERRLG=1.4  ERRHT=2.5
ALT=100.80  SATS=12  realsat=19  ERRLT=2.3  ERRLG=2.2  ERRHT=2.6
ALT=100.40  SATS=12  realsat=19  ERRLT=2.1  ERRLG=1.9  ERRHT=2.5
ALT=100.00  SATS=12  realsat=19  ERRLT=1.9  ERRLG=1.9  ERRHT=2.4
ALT=99.90  SATS=12  realsat=19  ERRLT=1.8  ERRLG=1.8  ERRHT=2.4
ALT=99.50  SATS=12  realsat=18  ERRLT=1.7  ERRLG=1.7  ERRHT=2.3
ALT=99.40  SATS=12  realsat=18  ERRLT=1.5  ERRLG=1.6  ERRHT=2.2
ALT=99.40  SATS=12  realsat=18  ERRLT=1.5  ERRLG=1.6  ERRHT=2.2
ALT=99.80  SATS=12  realsat=18  ERRLT=1.5  ERRLG=1.5  ERRHT=2.1
ALT=104.90  SATS=12  realsat=20  ERRLT=1.4  ERRLG=1.5  ERRHT=2.1
```

This is the broken example. Just replace "if (gps.time.isUpdated())" with "if (gps.altitude.isUpdated())" and it works:

```
#include <TinyGPS++.h>
TinyGPSPlus gps;

char data[] =
  "$GNRMC,003242.00,A,5121.22586,N,01159.45868,E,0.029,,030119,,,A,V*15\r\n"
  "$GNVTG,,T,,M,0.029,N,0.053,K,A*30\r\n"
  "$GNGNS,003242.00,5121.22586,N,01159.45868,E,AAAN,14,0.84,110.1,45.0,,,V*14\r\n"
  "$GNGGA,003242.00,5121.22586,N,01159.45868,E,1,12,0.84,110.1,M,45.0,M,,*4D\r\n"
  "$GNGSA,A,3,16,26,20,27,10,29,,,,,,,1.70,0.84,1.48,1*08\r\n"
  "$GNGSA,A,3,86,76,70,69,75,,,,,,,,1.70,0.84,1.48,2*00\r\n"
  "$GNGSA,A,3,01,08,07,,,,,,,,,,1.70,0.84,1.48,3*0A\r\n"
  "$GPGSV,4,1,14,07,09,334,,08,09,284,,10,20,166,47,13,02,048,,0*6D\r\n"
  "$GPGSV,4,2,14,15,04,081,38,16,67,269,28,20,41,145,44,21,65,070,,0*65\r\n"
[... 480 lines cut ...]
  Serial.println(F("5s"));
  delay(3000);
  Serial.println(F("Start"));

}

const char *p = data;

void loop()
{
  if (*p)
    gps.encode(*p++);

  if (gps.time.isUpdated())
  {

    Serial.print(F("ALT="));        Serial.print(gps.altitude.meters());
    Serial.print(F("  SATS="));     Serial.print(gps.satellites.value());
    Serial.print(F("  realsat="));  Serial.print(satnr.value());
    Serial.print(F("  ERRLT="));    Serial.print(errorlat.value());
    Serial.print(F("  ERRLG="));    Serial.print(errorlong.value());
    Serial.print(F("  ERRHT="));    Serial.println(errorheight.value());
  }
}
```

mikalhart, 2019-01-03:
Ah, at least I understand why this variant doesn't work.  Since you never query the value of gps.time, gpt.time.isUpdated() will return "true" always.  For every character read, you're printing out all the data, and that takes forever.  Add this like to the very last block:

```
    Serial.print(F("  TIME="));     Serial.println(gps.time.value());
```

Eheran1, 2019-01-03:
I think I know why it doesnt work after playing around a lot more:
The NMEA-informations come asynchronus and I have to trigger all my action after the LAST "gps.?????.isUpdated()" and not time or speed, since they are both befor the other values. Using the wrong "trigger" causes all sorts of absurd problems, even checksumm-errors. Which is the last item to be updated? In my case the last item is the height error at the end of GST. With that all 3 errors work.

The only thing that still doesnt work is 
```TinyGPSCustom satnr(gps, "GNGNS", 7);```
And that is far befor GST almost at the beginning of the NMEA-stream, right befor the (in my case incorrect) GGA-satellite value which ur library fetches.

This should return 18 and it does so in the simple Testsketch with fixed NMEA-Data.
$GNGNS,003451.00,5121.22645,N,01159.46189,E,DAAN,18,0.82,99.4,45.0,,0000,V*23

Can I stream the raw NMEA out somehow while its processed? The NMEA-Data I posted here was gathered over USB of the GPS unit, not via Serial of the µC.

Eheran1, 2019-01-03:
Not for me. Thanks for the great help and ur superb library.

closed 2019-01-03 by mikalhart