mikalhart/TinyGPSPlus #50: No GPS detected: check wiring.

StackerDEV, 2018-08-19:
Hi,

I am trying to test out my Neo6MV2. I have wired it like this;
https://www.bitsandparts.eu/blog/gps-module-ublox-neo6mv2-arduino-tutorial/

but it still says:
No GPS detected: check wiring.

I have also tried TinyGPS, NeoGPS. None of them seem to be able to get contact with my GPS module..
I have double checked the wiring, bad module?

AlyssonRowan, 2018-08-19:
> arduino-tutorial/

You might like to check that your GPS is configured correctly. I have one
that resets to binary mode when it has been powered off for a while.

*Alysson Rowan*

LinkedIn:      https://www.linkedin.com/in/alysson-rowan-68521692
Academia:    https://independentscholar.academia.edu/AlyssonRowan

StackerDEV, 2018-08-19:
@AlyssonRowan 
Is there any example out there that you recommend using? I mean a small guide/manual on all the steps to get it to work properly so to get started with this. I feel like I am missing a step somewhere. Thanks!

Fbo06, 2018-08-19:
Hello, did you cross the wire between Tx and Rx ?
GPS Tx  -> Arduino board pin RX
GPS Rx  -> Arduino board pin Tx

StackerDEV, 2018-08-26:
@mikalhart @Fbo06 
Both, yhanks got it working. I messed up on the RX/TX part. Gps RX must connect to TX on arduino and GPS TX to RX. I also found out that the vender that sold it to me listed it as 3.3 volt to power the module. When I connect it to 3.3volt with the voltage devider in place the gps module won't work properly. When I supplied 5v it worked like a skookum gps module. Got a fix in less than a minute.

[image]

mikalhart, 2018-08-26:
Great, thanks for sharing your results.

closed 2018-08-26 by mikalhart

callagga, 2018-09-07:
@mikalhart was there a run to run code against a GPS module connected to your quad via betaflight and the quad's mini-USB somehow?  Or do you need to get a "TDI USB to TTL Serial Converter" and connect the GPS module to this?

Steveiwonder, 2020-08-18:
> Thanks for the help @Fbo06 and @AlyssonRowan.
> 
> Yes, the most common error is connecting TX to TX. TX on the GPS side should be connected to RX on the Arduino side. This problem is exacerbated by the fact that some devices are labeled in reverse. Try both orientations.

this was the most helpful comment, i through a module away and got a second one thinking i screwed it during soldering. However, i was TX->TX and RX->RX ... but obviously i makes sense that TX from Arduino goes to RX on GPS and RX from GPS goes to RX on arduino!!

mohamedelbaradey, 2021-02-07:
> Hi,
> 
> I am trying to test out my Neo6MV2. I have wired it like this;
> https://www.bitsandparts.eu/blog/gps-module-ublox-neo6mv2-arduino-tutorial/
> 
> but it still says:
> No GPS detected: check wiring.
> 
> I have also tried TinyGPS, NeoGPS. None of them seem to be able to get contact with my GPS module..
> I have double checked the wiring, bad module? I have a Uno with a Atmega328P-PU on it and using the stock Arduino ide.

i have the same problem can anybody help me