mikalhart/TinyGPSPlus #52: Wrong satellite number when using custom fields

kostuch, 2018-08-22:
Not really a bug, but rather wrong logic in examples (SatElevTracker.ino and other using custom fields).
When satsInView.value() is not multiply of four (eg.11 or 6), then collected set of satellites has duplicates.
Instead of using:
    if (totalGPGSVMessages.isUpdated()) {
       for (int i=0; i<4; ++i)
       ...

Use loop:
for (int i = 0; currentMessage < totalMessages ? i < 4 : i < (atoi(satsInView.value()) % 4); ++i)

It will take care about last GPGSV sentence which can be "not complete" (read: less than 4 satellites)

Eheran1, 2018-12-31:
Another problem with satellite numbers:
My uBlox NEO-M8Q has different values of "numbers of satellites in use" in the NMEA sentences "GNGNS" and "GNGGA"... for whatever reason. GNGNS says the (correct) number, eg. 20, while GNGGA stays at a maximum of 12. Since GGA is the standard for TinyGPS++ I tryed to use:

"TinyGPSCustom satnr(gps, "GNGNS", 7);"

However, this doesnt work. I dont get any number out of this but rather "const char*" of unknown content. All my other custom values work just fine.

mikalhart, 2019-10-13:
Ah, indeed!  I'll leave this open and think about it a bit.

WizardTim, 2020-05-03:
@Eheran1, just tried to reproduce your problem with GNGNS SVs used (00-99) with an ESP8266 & UBlox NEO-6. I was able to get it to work without a problem. However, you (or others with similar problems) probably have enabled the message in u-center, but not saved it into flash. If I don’t save the configuration to flash after enabling the GNS message, the configuration is reset on power loss and no GNS message is transmitted thus I get no data in the value and isValid reports false.

Eheran1, 2020-05-05:
@WizardTim I dont see how this would matter. The "numbers of satellites in use" should be exactly the same for both the GAA and GNS sentence, no matter what setting anyone uses. But yes, I know about this and always save to flash - after all, U-Center warns about unsafed settings.

BTW: This code works now, but I dont know why (forgot) or if there is still a difference in the two sentences (havent checked since).
`TinyGPSCustom satnr(gps, "GNGNS", 7);`

WizardTim, 2020-05-05:
When the NMEA standard was created many years ago no one was expecting anyone could use more than 12 satellites for resolving location because the number of satellites visible at once was less than that, thus the GAA message was limited to 12 SVs. Now with our fancy multi-constellation receivers it’s very common to have a lot more than 12 SVs but those receivers have to be backwards compatible with older NMEA versions (currently 4.11) thus they limit the GAA SVs field to 12 but in the newer GNS field it can go up to 99. 

Good to hear you got it working 😊

Eheran1, 2020-05-05:
Very interesting! Thanks.

svdrummer, 2020-11-08:
Is this still open after a few years? Or resolved.

WizardTim, 2020-11-09:
kostuch's issue is about how GSV messages report data in chunks of 4 satellites where extra fields are left blank, their code suggestion would work (and might save a few clock cycles) but what they are talking about should already be handled by the line:

https://github.com/mikalhart/TinyGPSPlus/blob/cfb60e5e381c025ff7e52838897edab8a6e7d449/examples/SatelliteTracker/SatelliteTracker.ino#L99

@kostuch if your talking about something different or you've gotten actually bugged SV numbers please correct me. 

As for Eheran1's issue the problem is with the NMEA GAA message being limited to a maximum of 12 SVs due to the design of the very first GPS constellations. TinyGPSPlus can never fix this problem, instead people must either disable backwards compatibility for GAA messages or enable the newer GNS message which goes up to 99 SVs.

kostuch, 2020-11-09:
@WizardTim 
Issue I created has nothing to do with MAX_SATELLITES. It is related to "incomplete group" of satellites (4 in one nmea message).
As I mentioned, it's rather problem with user code used to fetch the data, not the library.