mikalhart/TinyGPSPlus #53: Change GPS baud

ajuancer, 2019-01-18:
Hi! I have a GPS module with 5 Pins: GND, VCC, Rx, Tx & PPS. I know you can define `static const uint32_t GPSBaud = 4800; 
`  but I would like to know if it's also available to use the PPS pin.

TD-er, 2019-01-18:
I still have to add a baud rate selector.
The functionality to use the PPS pin is built in, but it hasn't been tested yet.
Also make sure you do not connect the RX pin, or else there is a big chance the ESP will not boot properly after a restart.

ajuancer, 2019-01-18:
?

TD-er, 2019-01-18:
I did type a reply to it about what was just implemented in ESPeasy (my project, using this plugin) and what needs more testing.
Then I realized this wasn't the issue tracker of ESPeasy, so I removed my reply. (should have edited it)
So sorry for the confusion ;)

bamapookie, 2019-03-23:
The PPS pin is not a serial interface.  It doesn't have a baud setting.  In my experience (with the Adafruit GBS board) you read the PPS pin through one of your digital inputs.  As soon as it goes high, that is the start of the second.  It is independent of the GPS messages being sent serially over the TX/RX pins.

Hope that helps.

mikalhart, 2019-10-12:
FWIW, yes, you can change the baud rate in your sketch.

closed 2019-10-12 by mikalhart

sbonaime, 2020-10-12:
PPS means Pulse Per Second. It is one pulse every second which can validate date and time

TD-er, 2020-10-12:
N.B. the PPS pin on most modules only will output a signal (of 1.000000 Hz, probably even less than 1 usec jitter) if the GPS has either a fix or a sufficient reliable knowledge of time.