mikalhart/TinyGPSPlus #57: How to calculate Date and Time in different timezone from TinyGPSPlus class

AlexZibin, 2019-02-28:
Hi, 
Thank you for this great library.

Please advise the reliable way to calculate the Date and Time in different timezone.

**The example:**
**UTC:**  Date is 01.01.2019, Time is 23:59
Timezone is 2.
I want to receive **TZ2:** Date is 02.01.2019, Time is 01:59.

My guess is to get raw UTC_seconds, increase them by 24x60x60xTZ, and convert them back to date/time, but I can't find the proper places in your library.

Thank you in advance, I hope that if you will provide this hint, it will be helpful for many users.

PKLoops, 2019-03-09:
I have same problem, I need actual local time ( with or without daylight saving ) and local day, for TZ 10, East Coast Australia

Thank you too in advance

bamapookie, 2019-03-20:
You might want to look into the time and time zone libraries.
https://www.pjrc.com/teensy/td_libs_Time.html
https://github.com/JChristensen/Timezone

AlexZibin, 2019-03-21:
Wow! Thank you! 
The problem is solved!

closed 2019-03-21 by AlexZibin