mikalhart/TinyGPSPlus #61: How can I get a signal accuracy ?

GeorgeFlorian, 2019-06-06:
Hello !

I found an example online that uses this library and I can't quite understand what does the following line of code do.
```
gpsState.dist = TinyGPSPlus::distanceBetween(gps.location.lat(), gps.location.lng(), gpsState.originLat, gpsState.originLon);
```

I get that it calculates a difference in LAT and LNG, but for what purpose ?

**What should `gpsState.dist` represent ?**
Is it to calculate the signal accuracy ?

```
struct GpsDataState_t {
  double originLat = 0;
  double originLon = 0;
  double originAlt = 0;
  double distMax = 0;
  double dist = 0;
  double altMax = -999999;
  double altMin = 999999;
  double spdMax = 0;
  double prevDist = 0;
};
GpsDataState_t gpsState = {};
```

TD-er, 2019-06-06:
The HDOP value is some indicator on the accuracy of the received signal.
It is the standard deviation in meters. (99 if no fix)

mikalhart, 2019-10-13:
Thanks @TD-er.  @GeorgeFlorian, are we good?

closed 2019-10-13 by mikalhart