mikalhart/TinyGPSPlus #63: CustomFields When Blank

DrJaymz, 2019-06-27:
$PFLAA,0,-45322,54013,2709,1,406CCE!EZY29QP,23,,145,-10.7,9*70

Example where field is blank (the field after 1500).
$PFLAA,3,1500,,11,1,407076!GDFDO,,,,0.0,8*0B

When parsing using custom fields, when the field is blank I'm getting a value, which I think is whatever it was.  Updated is true, age is 1ms so it thinks its parsed it.  I specifically need to know when this field is blank for my application.

Am I being dumb here?

DrJaymz, 2019-06-28:
I tried for ages to get around this problem. So far no progress.

mikalhart, 2019-10-13:
Are you printing customvalue.value()?  I did a test program and it works as expected.  Sketch:
`#include "TinyGPS++.h"`

`const char *str1 = "$PFLAA,0,-45322,54013,2709,1,406CCE!EZY29QP,23,,145,-10.7,9*70\r\n";`
`const char *str2 = "$PFLAA,3,1500,,11,1,407076!GDFDO,,,,0.0,8*0B\r\n";`

`TinyGPSPlus gps;`
`TinyGPSCustom custom(gps, "PFLAA", 3);`
`void setup()`
`{`
`  Serial.begin(115200);`
`  Serial.println("Hello!");`
`  Serial.print("Before processing,value is:      "); Serial.println(custom.value());`
`  while (*str1) gps.encode(*str1++);`
`  Serial.print("After first sentence, value is:  "); Serial.println(custom.value());`
`  while (*str2) gps.encode(*str2++);`
`  Serial.print("After second sentence, value is: "); Serial.println(custom.value());`
`}`

`void loop() {`
`  // put your main code here, to run repeatedly:`

`}`

Output:
`Hello!`
`Before processing,value is:      `
`After first sentence, value is:  54013`
`After second sentence, value is:  `

I'll close this, but we can reopen if you have more trouble.

closed 2019-10-13 by mikalhart