mikalhart/TinyGPSPlus #65: latitude和longitude精度不够

liuyang77886, 2019-10-05:
hello，
怎么增加精度到6位

TD-er, 2019-10-05:
For those who (also) don't speak Chinese:

>Latitude and longitude are not accurate enough #65

> Hello,
> my current output: latitude":30.60,"longitude":104.13,
> how to increase the precision to 6 latitude":39.997761,"longitude":116.478935,

How do you convert the values to String?
The default number of decimals in `String` when converting a floating point is 2 decimals.
So you could convert it with something like this:
```c++
float lat = 123.4567891
String str(lat, 6);
```
It should then print something like this: `123.4567891`
Please note that a float only has 6 - 7 decimals precision, so you will loose some when you have more decimals before the "."

mikalhart, 2019-10-13:
@TD-er: right.
I usually do similar:
`Serial.print(latitude, 6);`

closed 2019-10-13 by mikalhart