mikalhart/TinyGPSPlus #68: Don't rely on arduino

feliwir, 2020-01-06:
Hello everyone,

why exactly does this library rely on arduino functionality? Mathematical functions could also be used from<cmath>. I'd love to use this great library with other microcontrollers.
Thanks

codygray, 2020-01-07:
I agree. There is no reason for this library to be Arduino-dependent. It is almost trivial to make it more universal. I have just submitted [a pull request](https://github.com/mikalhart/TinyGPSPlus/pull/69) that does exactly that. In the meantime, you are welcome to use [my fork](https://github.com/codygray/TinyGPSPlus/tree/universal).

mmaxxza, 2020-02-05:
Since @codygray 's fork did not compile for my board, we fixed tinygps plus by putting this in the start of the functions that depend on degrees and radians (courseto and distancefrom).
Else you can simply delete these functions if you are only logging or reporting backend.

```
#define _USE_MATH_DEFINES
#include <cmath>

#define PI           (M_PI)
#define HALF_PI      (M_PI / 2.0)
#define TWO_PI       (M_PI * 2.0)
#define DEG_TO_RAD   (M_PI / 180.0)
#define RAD_TO_DEG   (180.0 / M_PI)

#define radians(deg) ((deg)*DEG_TO_RAD)
#define degrees(rad) ((rad)*RAD_TO_DEG)
#define sq(x)        ((x)*(x))
```