mikalhart/TinyGPSPlus #78: lora issue

RustInPeaces, 2020-08-18:
Hi. 
First of all im pretty new at this so please bear with me.

So Im trying to get gps coordinates from a neo 6m gps to Arduino Uno to LoRa xl1278-smt, and sent to another LoRa xl1278-smt connected to a Arduino Nano.

I confirmed getting gps signals, and transimmer is sending packet, receiver is receiving packets, but all i get is Long 0. 0 deg and Lat 0.0 deg readings. Why? I dont get it, if it sends it should send the received gps signals right? Im also encoding the nmea to latitude/longitude before sending. Havent tried without yet.

Cheers :) 

Transmitter code:
//Code for Transmitter (Sender)

#include <SPI.h>
#include <LoRa.h>
#include <TinyGPS++.h>
#include <SoftwareSerial.h>

static const int RXPin = 4, TXPin = 3;
static const uint32_t GPSBaud = 9600;

//Create a TinyGPS++ object
TinyGPSPlus gps;

SoftwareSerial ss(RXPin, TXPin);
 
void setup() {

//Start Arduino hardware serial port at 9600 baud
  Serial.begin(9600);
//Start the software serial port at the GPS
  ss.begin(GPSBaud);
  while (!Serial);
  Serial.println("LoRa Sender");
  if (!LoRa.begin(433E6)) {
    Serial.println("Starting LoRa failed!");
    while (1);
  }
  
  LoRa.setTxPower(20);
}

void loop() {

  
  //This sketch displays information every time a new
  //sentence is correctly encoded.
  while (ss.available() >0){
   gps.encode(ss.read());
    if(gps.location.isUpdated()){
      Serial.print("Latitude= "); Serial.print(gps.location.lat(), 6);
      Serial.print(" Longitude= "); Serial.println(gps.location.lng(), 6);
    }

  //If valid location begin transmitting values.
  //Setup made ready for 16*2 LCD screen
  LoRa.beginPacket();
  LoRa.print("Latitude= "); LoRa.print(gps.location.lat(), 6);
  LoRa.print(" Longitude= "); LoRa.print(gps.location.lng(), 6);
  Serial.println("Sent via LoRa");
  LoRa.endPacket();
delay(1000);
  }
  }

Receiver code:

#include <SPI.h> 
#include <LoRa.h>

void setup() {
  //Start serial for debugging
  Serial.begin(9600);
  while (!Serial);

  Serial.println("LoRa Receiver");

//Set LoRa frequency and operation reply
if (!LoRa.begin(433E6)) { 
  Serial.println("Starting LoRa failed!");
  while (1);
}
}

void loop() {
//Try to parse packet
int packetSize = LoRa.parsePacket();
//If packet received
if(packetSize) {  
Serial.print("Received packet '");

//Read packet
while (LoRa.available()) {
 Serial.print((char)LoRa.read());

  }
}
  
}

TD-er, 2020-08-18:
`gps.location.isUpdated()` only shows when a string containing coordinates was processed.
It doesn't mean you have a fix.
Acquiring a fix may take quite some time initially, especially when being indoor and the GPS has no idea where it could be in the world.

RustInPeaces, 2020-08-18:
But the gps has a blue light when it has position, and that it has. Also i changed that to gps.location.isValid() , with same result.

mikalhart, 2021-07-13:
Your "delay(1000);" makes it all but certain that NMEA characters are being lost.  Rewrite this so that you send a packet only, say, when millis() - start > 1000.

Good rule of thumb: never use delay() in a sketch with TinyGPS.