mikalhart/TinyGPSPlus #80: GPRMC won't parse

Flupke1979, 2020-08-25:
Hello,
Hope you can help me.
I am using ai thinker A9G module.
TinyGPS++ only parses the GPGGA sentence when the GPS sends all sentences. I get LAT, LON and Height, but no speed or date.
When I process them 1 by 1 in a terminal program it works.

Sentence dump from GPS:
+GPSRD:$GPGGA,172356.000,5051.8430,N,00510.3105,E,1,7,1.23,32.6,M,47.4,M,,*62
$GPGSA,A,3,06,02,09,26,07,04,16,,,,,,1.53,1.23,0.90*01
$GPGSV,3,1,12,09,71,066,14,07,51,161,18,02,45,280,20,06,45,214,16*72
$GPGSV,3,2,12,04,32,068,31,37,29,159,,30,25,187,,05,22,299,13*7F
$GPGSV,3,3,12,16,15,060,33,29,09,335,,26,07,025,18,03,02,120,15*73
$GPRMC,172356.000,A,5051.8430,N,00510.3105,E,0.188,11.19,250820,,,A*63
$GPVTG,11.19,T,,M,0.188,N,0.348,K,A*0B

Do you have any idea? 

Thanks in advance.
Kim

TD-er, 2020-08-26:
Not sure what kind of NMEA dialect that is, but I cannot match it with the description [here](http://aprs.gids.nl/nmea/#rmc)
It looks like the checksum at the end of the line doesn't have a * followed by the checksum, but in case of the GPRMC example line you gave it has an A.
The GPVTG does seem to have a combination of both.

So simply stated, it looks like the checksum does fail here as it is not being recognized as a checksum field.

Flupke1979, 2020-08-26:
Thanks.
I did a workaround by connecting the A9G secondary GPS-only output to an extra RX input.
This output sends proper sentences, that are parsed ok.
Strange that the main A9G output sends wrong messages.

closed 2020-08-26 by Flupke1979