mikalhart/TinyGPSPlus #82: Data is not getting generated here as you can see in the attachment, need your help please

NilothpalBhattacharya, 2020-10-05:
Hi, I am facing this issue , not sure if I am even able to fetch the correct data here. Please can you help here. 

[image]

mikalhart, 2020-10-05:
I think the UBlox baud rate should be set to 9600.

ss.begin(9600);

NilothpalBhattacharya, 2020-10-06:
Thanks for coming back. Yes I have set the correct Baud rate . The default is 9600 for Neo 6M I hope. Is it okay if I can give you a call today please? Thankyou

TD-er, 2020-10-06:
Just a simple question as I don't see it in your screenshot.
What pins on the Arduino board are assigned to the serial port you intent to use?
And remember to "cross" the RX and TX.
So TX on the GPS should be connected to RX on the serial port (the other one doesn't matter as isn't used)

Maybe for testing, you can also print every character you receive to help diagnose what's wrong here.

mikalhart, 2020-10-06:
You can see in the screenshot on the right that large numbers of characters are being received, without any valid sentences and with lots of chksum fails.  This suggests that the wiring is correct but the baud rate is wrong.  Indeed, we can see in the left screen that the baud rate is set to 4800.

Agree with TD-er that printing out each character would be useful.

NilothpalBhattacharya, 2020-10-13:
Sorry guys yesterday took the module outside under clear sky,it was working.just had to ensure the connection of pins were correct and the baud rate. Thankyou for your support.

NilothpalBhattacharya, 2020-10-13:
I actually used Arduino Nano to interface all my sensors and Gps and LoRA module. So all that worked fine but it is an interesting work to learn from I realized.

NilothpalBhattacharya, 2020-10-13:
> Just a simple question as I don't see it in your screenshot.
> What pins on the Arduino board are assigned to the serial port you intent to use?
> And remember to "cross" the RX and TX.
> So TX on the GPS [...]

I was using pin 7 as Rx, 6 as Tx of Arduino Nano

svdrummer, 2021-04-18:
Should this be changed to CLOSED?

closed 2021-07-13 by mikalhart