mikalhart/TinyGPSPlus #84: i can't get GPS data when i try to send it via LoRa

Mohamedx7, 2020-11-03:
I'm using TTGO T BEAM V1 board first, when I run GPS code it runs perfectly but when I add LoRa to the code to send to another TTGO T BEAM board it shows 0.00 for both latitude and longitude .... any help??

`#include <Wire.h>
#include <axp20x.h>
#include <TinyGPS++.h>
#include "SSD1306.h" 
#include "images.h"
#include <SPI.h>
#include <LoRa.h>

#define SCK     5    // GPIO5  -- SX1278's SCK
#define MISO    19   // GPIO19 -- SX1278's MISnO
#define MOSI    27   // GPIO27 -- SX1278's MOSI
#define SS      18   // GPIO18 -- SX1278's CS
#define RST     14   // GPIO14 -- SX1278's RESET
#define DI0     26   // GPIO26 -- SX1278's IRQ(Interrupt Request)
#define BAND  868E6

SSD1306 display(0x3c, 21, 22);
String rssi = "RSSI --";
String packSize = "--";
String packet ;

#define UBLOX_GPS_OBJECT()  TinyGPSPlus gps
#define GPS_BAUD_RATE 9600
#define GPS_RX_PIN 34
#define GPS_TX_PIN 12

#define I2C_SDA 21
#define I2C_SCL 22
#define AXP192_SLAVE_ADDRESS 0x34

UBLOX_GPS_OBJECT();

AXP20X_Class axp;

bool axp192_found = false;

char buff[5][256];
uint64_t gpsSec = 0;

void scanI2Cdevice(void)
{
    byte err, addr;
    int nDevices = 0;
    for (addr = 1; addr < 127; addr++) {
        Wire.beginTransmission(addr);
        err = Wire.endTransmission();
        if (err == 0) {
            Serial.print("I2C device found at address 0x");
            if (addr < 16)
                Serial.print("0");
            Serial.print(addr, HEX);
            Serial.println(" !");
            nDevices++;

            if (addr == AXP192_SLAVE_ADDRESS) {
                axp192_found = true;
                Serial.println("axp192 PMU found");
            }
        } else if (err == 4) {
            Serial.print("Unknow error at address 0x");
            if (addr < 16)
                Serial.print("0");
            Serial.println(addr, HEX);
        }
    }
    if (nDevices == 0)
        Serial.println("No I2C devices found\n");
    else
        Serial.println("done\n");
}

void setup() {
  
  pinMode(16,OUTPUT);
  pinMode(2,OUTPUT);

  digitalWrite(16, LOW);    // set GPIO16 low to reset OLED
  delay(50); 
  digitalWrite(16, HIGH); // while OLED is running, must set GPIO16 in high

  while (!Serial);
  Serial.println();
  Serial.println("LoRa Sender Test");
  
  SPI.begin(SCK,MISO,MOSI,SS);
  LoRa.setPins(SS,RST,DI0);
  if (!LoRa.begin(868E6)) {
    Serial.println("Starting LoRa failed!");
    while (1);
  }
  //LoRa.onReceive(cbk);
//  LoRa.receive();
  Serial.println("init ok");
  display.init();
  display.flipScreenVertically();  
  display.setFont(ArialMT_Plain_10);
   
  delay(1500);
  
  
  
  
  
  
  
  
  
  Serial.begin(115200);

  delay(5000);
  
  Wire.begin(I2C_SDA, I2C_SCL);
  
  scanI2Cdevice();
  
  if (axp192_found) {
      if (!axp.begin(Wire, AXP192_SLAVE_ADDRESS)) {
          Serial.println("AXP192 Begin PASS");
          // power on ESP32 & GPS
          axp.setPowerOutPut(AXP192_LDO3, AXP202_ON);
          axp.setPowerOutPut(AXP192_DCDC1, AXP202_ON);
          axp.setDCDC1Voltage(3300);  //esp32 core VDD    3v3
          axp.setLDO3Voltage(3300);   //GPS VDD      3v3

          Serial.printf("DCDC1: %s\n", axp.isDCDC1Enable() ? "ENABLE" : "DISABLE");
          Serial.printf("DCDC2: %s\n", axp.isDCDC2Enable() ? "ENABLE" : "DISABLE");
          Serial.printf("LDO2: %s\n", axp.isLDO2Enable() ? "ENABLE" : "DISABLE");
          Serial.printf("LDO3: %s\n", axp.isLDO3Enable() ? "ENABLE" : "DISABLE");
          Serial.printf("DCDC3: %s\n", axp.isDCDC3Enable() ? "ENABLE" : "DISABLE");
          Serial.printf("Exten: %s\n", axp.isExtenEnable() ? "ENABLE" : "DISABLE");
               
          // Set mode of blue onboard LED (OFF, ON, Blinking 1Hz, Blinking 4 Hz)
          // axp.setChgLEDMode(AXP20X_LED_OFF);
          //axp.setChgLEDMode(AXP20X_LED_LOW_LEVEL);
          axp.setChgLEDMode(AXP20X_LED_BLINK_1HZ);
          //axp.setChgLEDMode(AXP20X_LED_BLINK_4HZ);
      } else {
          Serial.println("AXP192 Begin FAIL");
      }
      
  } else {
      Serial.println("AXP192 not found");
  }
      
  Serial1.begin(GPS_BAUD_RATE, SERIAL_8N1, GPS_RX_PIN, GPS_TX_PIN);

 display.init();
  display.flipScreenVertically();  
  display.setFont(ArialMT_Plain_10);
   
  delay(1500);
  
}

void loop() {

display.clear();
  display.setTextAlignment(TEXT_ALIGN_LEFT);
  display.setFont(ArialMT_Plain_10);
  
  display.drawString(0, 0, "Latitude: ");
  display.drawString(60, 0, String(gps.location.lat()));

  display.drawString(0, 10, "Longitude: ");
  display.drawString(60, 10, String (gps.location.lng()));

  display.drawString(0, 20, "satellites: ");
  display.drawString(60, 20, String(gps.satellites.value()));

  display.drawString(0, 30, "UTC: ");
  
  display.drawString(60, 30, String(gps.time.hour()));
  display.drawString(70, 30, " : ");

  display.drawString(80, 30, String (gps.time.minute()));
  display.drawString(90, 30, " : ");

  display.drawString(100, 30, String (gps.time.second()));
  display.drawString(0, 30, " : ");

  display.drawString(7, 50, " Designed by Mohamed");

  
  display.display();

 // send packet
  LoRa.beginPacket();
  
  LoRa.print("Latitude:");
  LoRa.print(gps.location.lat());

  LoRa.print("\nLongitude:");
  LoRa.print(gps.location.lng());

  LoRa.print("\nsatellites: ");
  LoRa.print(gps.satellites.value());

  //LoRa.print("counter ");
  //LoRa.print(counter);

  
  LoRa.endPacket();

  
  // put your main code here, to run repeatedly
  static uint64_t gpsMap = 0;

  while (Serial1.available())
  gps.encode(Serial1.read());

  if (millis() > 5000 && gps.charsProcessed() < 10) {
    snprintf(buff[0], sizeof(buff[0]), "T-Beam GPS");
    snprintf(buff[1], sizeof(buff[1]), "No GPS detected");
    Serial.println(buff[1]);
    return;
  }
  if (!gps.location.isValid()) {
    if (millis() - gpsMap > 1000) {
        snprintf(buff[0], sizeof(buff[0]), "T-Beam GPS");
        snprintf(buff[1], sizeof(buff[1]), "Positioning(%llu)", gpsSec++);
        Serial.println(buff[1]);
        gpsMap = millis();
    }
  } else {
    if (millis() - gpsMap > 1000) {
        snprintf(buff[0], sizeof(buff[0]), "UTC:%d:%d:%d", gps.time.hour(), gps.time.minute(), gps.time.second());
        snprintf(buff[1], sizeof(buff[1]), "LNG:%.4f", gps.location.lng());
        snprintf(buff[2], sizeof(buff[2]), "LAT:%.4f", gps.location.lat());
        snprintf(buff[3], sizeof(buff[3]), "satellites:%u", gps.satellites.value());
        Serial.println(buff[0]);
        Serial.println(buff[1]);
        Serial.println(buff[2]);
        Serial.println(buff[3]);
        gpsMap = millis();
    }
  }
}`

TD-er, 2020-11-03:
Please use 3 back ticks wrapping your code.

```
Like this at the start on a single line: ```c++
at the end of the code block on a new line: ```
```

Mohamedx7, 2020-11-04:
thank you for replying to me .... it  worked

closed 2020-11-04 by Mohamedx7