mikalhart/TinyGPSPlus #85: Add to Arduino Library Manager

santaimpersonator, 2021-02-05:
Is there a reason why this hasn't been submitted to be added into the Arduino library manager yet?

santaimpersonator, 2021-06-17:
@mikalhart Do you mind if I submit a pull request in the [Arduino library repo](https://github.com/arduino/library-registry) to include your library?

mikalhart, 2021-06-17:
That would be lovely.  Thank you @santaimpersonator.  I was just thinking today that I ought to do that.

svdrummer, 2021-06-19:
Make sure it say TinyGps++ as there is a tinygps library on there at the moment.

santaimpersonator, 2021-06-24:
I created a [pull request](https://github.com/arduino/library-registry/pull/110) to add this library to the Arduino Library Manager. Unfortunately, there was an error:

```
Linting library in TinyGPSPlus
Rule LS008 result: fail
INFO: No header file found matching library name (TinyGPS__.h). Best practices are for primary header filename to match library name.
Rule LP011 result: fail
ERROR: Prohibited character(s) in library.properties name value: TinyGPS++. See: https://arduino.github.io/arduino-cli/latest/library-specification/#libraryproperties-file-format
Rule LD002 result: fail
WARNING: No license file found. See: https://docs.github.com/en/free-pro-team@latest/github/creating-cloning-an
```

I don't think the linter can recognize certain special characters and Arduino doesn't like the library name: `TinyGPS++`. To comply with their specifications:
> Library names must contain only basic letters (A-Z or a-z) and numbers (0-9), spaces ( ), underscores (_), dots (.) and dashes (-). 

Would you be opposed to renaming the [library](https://github.com/mikalhart/TinyGPSPlus/blob/master/library.properties#L1) and [files](https://github.com/mikalhart/TinyGPSPlus/tree/master/src) to `TinyGPSPlus`, `TinyGPSPlusPlus`, or etc? Obviously, this also requires a version bump and new release.
- I can create a pull request with the name changes. I was thinking `TinyGPSPlus`, to align with the repository name.

santaimpersonator, 2021-06-30:
I just created a pull request for the mentioned changes: https://github.com/mikalhart/TinyGPSPlus/pull/92

I went with `TinyGPSPlus`, to align with the repository name.

santaimpersonator, 2022-01-05:
I closed https://github.com/mikalhart/TinyGPSPlus/pull/92 and https://github.com/arduino/library-registry/pull/110. The desired library name has been taken, by a fork of this repository. See details here: https://github.com/mikalhart/TinyGPSPlus/pull/92#issuecomment-1005408105

I can reopen https://github.com/arduino/library-registry/pull/110 and submit a new pull request, once a new library name has been decided on or get the forked version of this library (https://github.com/Tinyu-Zhao/TinyGPSPlus) modified... removed or name changed.

per1234, 2022-01-05:
Thanks for your efforts to add this valuable library to the Arduino Library Manager @santaimpersonator.

I remain available to answer any questions regarding Library Manager and the submission process.

I suppose we can be happy that the library has finally been added, even though not from the canonical source. It is a bit unfortunate that it is now specified as only being compatible with the ESP32 architecture:
https://github.com/Tinyu-Zhao/TinyGPSPlus/blob/310ff90216b2ec2d30f719fccff38bd6921ffe40/library.properties#L9
```
architectures=esp32
```

mikalhart, 2022-01-05:
I would like to second @per1234's thanks to @santaimpersonator.  I will try to contact the owner of the forked repo to see if he can rename it to TinyGPSPlus-ESP32 or something.  Apologies for the slowness on my end.

TD-er, 2022-01-05:
Maybe you can also get his attention by simply naming the user: @Tinyu-Zhao
That makes explaining the issue a lot easier ;)

mikalhart, 2022-01-05:
Heh, quite.

Hey @Tinyu-Zhao,

Would it be possible to rename the fork of my library that you contributed to the Arduino Library Manager to **TinyGPSPlus-ESP32** or something similar?  We are hoping to submit the parent repo under the name **TinyGPSPlus**.  (See thread above.)

Thank you very much,

Mikal Hart

Tinyu-Zhao, 2022-01-09:
> 嘿，相当。
> 
> 嘿 @Tinyu-Zhao，
> 
> 是否可以将您向Arduino图书馆管理器贡献的我的图书馆分叉重命名为**TinyGPSPlus-ESP32**或类似的东西？我们希望以**TinyGPSPlus**的名义提交父回购协议。（见上面的线程。）
> 
> 非常感谢，
> 
> 米卡尔·哈特

> Heh, quite.
>
> Hey @Tinyu-Zhao,
>
> Would it be possible to rename the fork of my library that you contributed to the Arduino Library Manager to **TinyGPSPlus-ESP32** or something similar? We are hoping [...]

Of course you can, thanks for your repository!

per1234, 2022-01-09:
OK. So here is the list of things that need to be done to get this library into Library Manager:

- [x] Update the `name` value in the `library.properties` metadata file of [`Tinyu-Zhao/TinyGPSPlus`](https://github.com/Tinyu-Zhao/TinyGPSPlus)
  - DONE: https://github.com/Tinyu-Zhao/TinyGPSPlus/commit/e45ca4f8aacae83bc79ae67d387f5b87c7c570cb
- [x] Submit PR to update the name in the registration data for `Tinyu-Zhao/TinyGPSPlus`
  - DONE: https://github.com/arduino/library-registry/pull/878
- [ ] Wait for Arduino's Library Manager database maintainer to update the name data in the database.
    They will merge https://github.com/arduino/library-registry/pull/878 when that has been done so you can subscribe to that PR to get a notification.
- [ ] Merge https://github.com/mikalhart/TinyGPSPlus/pull/92
- [ ] Create a release or tag in `mikalhart/TinyGPSPlus`
- [ ] Reopen https://github.com/arduino/library-registry/pull/110 and trigger another check by the bot
   (or open another PR if you prefer, either is perfectly fine)

mikalhart, 2022-01-10:
@per1234 - Fantastic.  Thanks for making the bullet list.
@Tinyu-Zhao - Thanks so much for your cooperation.

Do we need to re-open PR #92?

santaimpersonator, 2022-01-10:
I just reopened pull request https://github.com/mikalhart/TinyGPSPlus/pull/92

* The version number is currently set at 1.0.3: https://github.com/santaimpersonator/TinyGPSPlus/commit/14cf11488c717a62ccccffdf4c42dc9a9bd168fe
* @mikalhart Let me know is you'd like the version number to be modified for the release/tag

per1234, 2022-01-19:
Good news everyone! https://github.com/arduino/library-registry/pull/878 has now been merged, so one more item has been completed on the checklist.

santaimpersonator, 2022-01-19:
@per1234
I was just about to comment the same

@mikalhart
I think the next step is to merge [Arduino library compatibility #92](https://github.com/mikalhart/TinyGPSPlus/pull/92). As soon as the merge is completed and the release is created, I can reopen [Add TinyGPSPlus arduino/library-registry#110](https://github.com/arduino/library-registry/pull/110) to trigger another check

santaimpersonator, 2022-01-20:
@mikalhart Thanks for merging the pull request (#92). Would you mind also creating a release with the version number `1.0.3` too? Thanks!

santaimpersonator, 2022-01-24:
@per1234 Could I reopen the pull request to the Arduino library without a new version release or will that throw an error for the bot? (i.e. Does the version of the latest release have to match what's in the [`libraries.properties` file](https://github.com/mikalhart/TinyGPSPlus/blob/master/library.properties#L2)?)

per1234, 2022-01-24:
>Could I reopen the pull request to the Arduino library without a new version release

You are welcome to reopen the pull request, but it will only be merged if, at the time of the tag, the repository was compliant with all the requirements listed here:
https://github.com/arduino/library-registry/blob/main/FAQ.md#submission-requirements

The bot and the indexing system only look at tags. These are the units of release for the Arduino Library Manager. So even if the library is 100% compliant at the tip of the `master` branch, the automated systems will not see that. This is what it will see:

https://github.com/mikalhart/TinyGPSPlus/tree/v1.02b

>Does the version of the latest release have to match what's in the `libraries.properties` file?

It is best practices for the tag name to match the value of the `version` field in `library.properties`, but that is only to avoid confusion for humans. The automated systems don't care at all what the tag name is. For them, the tag is only a marker at a point in the repository.

All versioning in Library Manager is done using the value of the `version` field in `library.properties`.

You do need to make sure to remember to update this field before creating each new release tag. The reason is that Library Manager will ignore any tag that has a `version` field value matching (in [semver](https://semver.org/) form) one of the releases of the library that are already in Library Manager. This is the most common cause of library releases being rejected by the Library Manager system.

mikalhart, 2022-01-24:
Minor issue: GitHub doesn't consider a string like "1.0.3" to be a valid tag.  It suggests instead "v1.0.3", which I'm happy to do... and that's consistent with previous releases... except that that doesn't strictly match the string library.properties.  Is that a problem?

per1234, 2022-01-24:
It is not a problem at all. You are welcome to use any tag name format you like. The Library Manager system doesn't care at all. 

There are requirements for the format of the value of `version` field of `library.properties` though, so make sure to comply with those whenever you update the `version` field:

https://arduino.github.io/arduino-cli/latest/library-specification/#libraryproperties-file-format

The current `version` field value (1.0.3) is compliant, so no worries there.

santaimpersonator, 2022-01-24:
Thanks for the insight and creating the new release, I'll reopen https://github.com/arduino/library-registry/pull/110

santaimpersonator, 2022-01-24:
Got https://github.com/arduino/library-registry/pull/110 merged!

- Had to fetch the upstream commits in my fork and retrigger the bot, but everything worked
- Thank you to @per1234 and @mikalhart for all your help and work
- I'll close the issue

closed 2022-01-24 by santaimpersonator