mikalhart/TinyGPSPlus #91: Tiny gps not reading

Max1092, 2021-06-16:
hello I am attempting to use the simple test example on the tiny gps library but I am having an issue with it working. I am using a mega 2560 with a Neo 6m gps module. I have the rx and tx pins connected to 19 and 18 respective but whenever I try to pull data on the serial monitor I just get this error 
** No characters received from GPS: check wiring **
 CHARS=0 SENTENCES=0 CSUM ERR=0
I have triple checked the wiring and its correct along with the baud rate, I changed the code so pins 18 and 19 are correct and the baud rate to 9600 but I can't seem to get it to work, I'm hoping you might have some insight as I have been trying to get this to work for weeks now and cannot figure it out. thanks

mikalhart, 2021-06-16:
Hi.

 

99% of errors like this are either

a.	Wiring mismatch.  Try hooking GPS.TX to Arduino.RX and GPS.RX to Arduino.TX.  (If you’re already doing that, try it the other way: TX-TX, RX-RX).
b.	Using wrong serial port in code.  You should be using Serial1 instead of SoftwareSerial if you’re using the Mega 2560’s actual (second) serial port.

 

Hope this helps…!!

 

Mikal

Jorin-Post, 2021-07-09:
If you have a new GPS model it takes some time to get data.
I let it run for a half hour and still not all the raw data was filled.
I hope you already figured it out.

mikalhart, 2021-07-13:
@Jorin-Post Quite right, but well before you get a fix you would expect to see characters flowing immediately.  @Max1092 reports "CHARS=0".  Actually on closer reading I see that Max's wiring is likely wrong: "rx and tx pins connected to 19 and 18".  Should be the other way around.

closed 2021-07-13 by mikalhart