mikalhart/TinyGPSPlus #95: Fake Neo 6m modules

orimate, 2021-07-13:
Hi, 

this not an issue it can be closed instantly. I wanted to share the result of my experince. After I spended 2 days of my life to get work the modules, I landed in a forum where I read that there is a lot of fakes on the markt. I read out the hardware version and as you can see it was UBX G5-xxx. After that I checked the label on the chip and I pulled it down and it is really an unlabelled version of 5 modul. What can I say? I write the post because if you read this and I can save time for you, than it worth it. 

[image]
[image]

TD-er, 2021-07-13:
Another way to know you're not dealing with a real one is trying to send data to it.
A number of units I've seen did not accept anything from the serial port.

mikalhart, 2021-07-13:
Thank you for sharing.  I do like those modules you've photographed.  Is there are particular vendor we should steer clear of?

orimate, 2021-07-13:
They were bought from ebay from different seller. I contacted with on of them, they don't know how can be happened. (Not chinese seller, of course they are saying we don't know) The seller said also this chip is one of the charge what are arrived on the last week. We have two problems. The first one is the chip can be completly chinese or the second one which is my case the chip is probably genuie but a relabaled old one. They are not able to set your configuration. Basically ebay or aliexpress or something like this portals are not safe. I have no idea where could I get a good one. I guess the problem is not new one. I founded an aritcle from the vendor from 2009 how can the user proof that chips are good or not. Here is couple of discussion in this  topic.

[link1](https://portal.u-blox.com/s/question/0D52p00008cihGk/fake-or-real-neo6m)
[link2](https://portal.u-blox.com/s/question/0D52p00008HKmpICAT/real-ublox-or-fake-chinese-)
[link3](https://portal.u-blox.com/s/question/0D52p00008HKEEECA5/psa-fake-ublox-modules-and-potential-ways-to-identify-them)

I hope that it helps.

TD-er, 2021-07-14:
Not only the modules can be counterfeit, but also the ceramic antennas.
Some of the "fake" modules are actually working fine. They are not uBlox, you may not be able to even set the baud rate etc., but they work just fine. 
But I do have a bag full of ceramic antennas, the bigger ones, which are just fake.
The small ceramic ones (rectangular ones) do work fine (not great sensitivity), but the 25x25mm ones I have here do not allow to get a single fix. Not even possible to get valid time.

hpsaturn, 2022-10-06:
I found another fuck seller that sell fake or refurbish modules:

[Aliexpress](https://www.aliexpress.com/item/1005002904682869.html)

https://user-images.githubusercontent.com/423856/194248202-90b7ac32-504d-4d64-9a1f-ff718850c04f.mp4