mikalhart/TinyGPSPlus #96: encode always returns false

ErwinVorenhout, 2021-07-29:
When trying to use DeviceExample with Arduino nano , using software serial on pin 8 , 7 encoding
always returns false. when using the basic example it seems to be running just fine.

**_What i've tried so far:_**

I tried if the ss.read() gets any valid NMEA data, which seem to be the case.

Any ideas?

ErwinVorenhout, 2021-07-29:
`#include <TinyGPS++.h>
#include <SoftwareSerial.h>

const char *gpsStream = "$TESTI,3,1,2,1,2,0,2,2,75*72\r\n";
static const int RXPin = 8, TXPin = 7;
static const uint32_t GPSBaud = 19200;

// The TinyGPS++ object
TinyGPSPlus gps;
TinyGPSCustom test(gps, "TESTI", 9); // $GPGSA sentence, 17th element
SoftwareSerial ss(RXPin, TXPin);

void setup()
{
  Serial.begin(19200);
  ss.begin(GPSBaud);

  // works
  while (*gpsStream)
    if (gps.encode(*gpsStream++))
      displayInfo();
      
  Serial.println(F("Done."));
}

void loop()
{ 
  //doesn't work when sending data
  while (ss.available() > 0){
    if (gps.encode(ss.read()))
      displayInfo();
  }
}

void displayInfo()
{
  Serial.print(test.value());
  Serial.println();
}`

ErwinVorenhout, 2021-07-29:
```#include <TinyGPS++.h>
#include <SoftwareSerial.h>

const char *gpsStream = "$TESTI,3,1,2,1,2,0,2,2,75*72\r\n";
static const int RXPin = 8, TXPin = 7;
static const uint32_t GPSBaud = 19200;

// The TinyGPS++ object
TinyGPSPlus gps;
TinyGPSCustom test(gps, "TESTI", 9); // $GPGSA sentence, 17th element
SoftwareSerial ss(RXPin, TXPin);

void setup()
{
  Serial.begin(19200);
  ss.begin(GPSBaud);

  // works
  while (*gpsStream)
    if (gps.encode(*gpsStream++))
      displayInfo();
      
  Serial.println(F("Done."));
}

void loop()
{ 
  //doesn't work when sending data
  while (ss.available() > 0){
    if (gps.encode(ss.read()))
      displayInfo();
  }
}

void displayInfo()
{
  Serial.print(test.value());
  Serial.println();
}```

mikalhart, 2021-08-24:
Hi @preshantram.  If encode never returns true, it's almost certainly (a) not getting any characters [usually crossed wires] or (b) getting _bad_ characters [usually incorrect baud rate].  I'd suspect the latter.  19200 is a rather unusual baud rate in my experience.

M

preshantram, 2021-08-24:
It was due missing CR, LF in the stream.

closed 2022-01-05 by mikalhart