mikalhart/TinyGPSPlus #98: problem parsing longitude

jensvanhoof86, 2021-11-01:
Hi,

I've got a strange problem with the library. For some reason, I get back a value of 90° longitude instead of the expected 9°.
Here's the bare minimum code that still produces the problem:

```
#include``` <TinyGPS++.h>
TinyGPSPlus gps;
double lon;

void setup() {
Serial.begin(19200);

}
void loop() {
while(Serial.available()>0){
if (gps.encode(Serial.read())){
DataParsed();
}}}

void DataParsed(){
lon = gps.location.lng();
Serial.println(lon);
}
```

The input phrase is:
$GPGGA,060054,4032.6766,N,009007.1100,E,2,05,1.5,6766.0,M,0.0,M,,*7C

And I'm getting back 90.12 as a result.
Any ideas? :-)

Thanks!

jensvanhoof86, 2021-11-01:
Can't believe I missed the extra zero after the nine..
One of the reasons I didn't see it is because I tried out the phrase here: http://freenmea.net/decoder, and it ignored that extra zero.

mikalhart, 2022-01-05:
Hi @jensvanhoof86,

So... I should close this, right?

Thanks,

M

closed 2022-01-24 by mikalhart