mikalhart/TinyGPSPlus #100: How to get the correct time of my time zone

Oscar-Chung-CKL, 2022-03-24:
i have download the library and i am using the neo-6m module on Arduino

I am able to get the time and the gps location, but the time is not the time in my timezone, may i know how can i solve the issue so that i can get the correct time of my time zone ?

Jens869, 2022-10-17:
`
#include <TinyGPS++.h>                                      // TinyGPS++  V 1.0.2 or higher   
#include <Time.h>                                               // Time V1.6.1 or higher
#include <Timezone.h>                                       // Timezone V1.2.4 or higher
 
TimeChangeRule CEST = {"", Last, Sun, Mar, 2, 120};      // Europa timezone
TimeChangeRule CET  = {"", Last, Sun, Oct, 3, 60}; 
Timezone CE(CEST, CET);
TimeChangeRule *tcr;  

CODE .... CODE

//if (gps.time.isValid())  // BUG in TinyGPS++
      if (gps.date.month()>0)
      {
        
        setTime(gps.time.hour() , gps.time.minute(), gps.time.second() , gps.date.day(), gps.date.month(), gps.date.year());
        time_t cet=CE.toLocal(now(),&tcr);          // Zeiger auf Zeitzone
        setTime(cet);                               // auf MEZ / MESZ umschalten   
      
       }
`

JasonPittenger, 2023-10-27:
NMEA sentence nZDA is supposed to give you time zone information.

 information: ZDA
 Descripiton: Time and date information.
 Type:		Output
 Format: 		$--ZDA,UTCtime,day,month,year,ltzh,ltzn*CS<CR><LF>
 Example: 	$GPZDA,235316.000,02,07,2011,00,00*51
 Field 	name 		Format 				Parameter Description
 1 		$--ZDA 		String 				Message ID, ZDA statement header,'--' is the system identifier
 2 		UTCtime 	hhmmss.sss 			UTC time when positioning
 3 		day 		Numerical value 	Day, fixed two digits, value range 01~31
 4 		month 		Numerical value 	Month, fixed two digits, value range 01~12
 5 		year 		Numerical value 	Year, fixed four digits
 6 		ltzh 		Numerical value 	This time zone is hour
 7 		ltzn		Numerical value 	Minutes in this time zone
 8 		CS 			Hexadecimal value 	Checksum, the exclusive OR of all characters between $ and * (not including $ and *)
 9 		<CR><LF> 	character 			Carriage return and line feed
 
Use the following to extract the information.
```
TinyGPSCustom	LocalTimeZoneHour(gps, "GPZDA", 5);
TinyGPSCustom	LocalTimeZoneMinute(gps, "GPZDA", 6);
```
Those will give you the offset in hours and minutes from UTC

Jens869, 2023-10-27:
after a year, what are you talking about?

JasonPittenger, 2023-10-27:
I know it's a year later, but the internet is forever.  
Someone else is bound to have a similar question.
I was unaware until today that the NMEA sentence GPZDA has the local timezone offset in it.
I plan to use this myself.

syproduction, 2023-11-27:
To make it work you need to use "whole constellation", thus use command GNZDA
```
TinyGPSCustom zHour(gps, "GNZDA", 5);
TinyGPSCustom zMinute(gps, "GNZDA", 6);
```
Anyway in my case it outputs 00:00 although I'm in GMT+1

Jason-Pittenger, 2023-11-27:
> Anyway in my case it outputs 00:00 although I'm in GMT+1

That's what I got too.  I think only certain GPS devices can interpret the offset information, and the rest output 0 all the time.