mikalhart/TinyGPSPlus #101: Wrong in date

Firered97, 2022-04-28:
Hey, I need help regarding date from the GPS. Right now my GPS is function as I can get my coordinate and time function also functioning as well but there is a problem with the date as it display not according to current date as shown on the image below.

[image]

 `  void displayInfo()
{
        // get time from GPS module
      if (gps.time.isValid())
      {
        Minute = gps.time.minute();
        Second = gps.time.second();
        Hour   = gps.time.hour();
      }
 
      // get date drom GPS module
      if (gps.date.isValid())
      {
        Day   = gps.date.day();
        Month = gps.date.month();
        Year  = gps.date.year();
      }
 
      if(last_second != gps.time.second())  // if time has changed
      {
        last_second = gps.time.second();
 
        // set current UTC time
        setTime(Hour, Minute, Second, Day, Month, Year);
        // add the offset to get local time
        adjustTime(time_offset);
 
        // update time array
        Time[12] = second() / 10 + '0';
        Time[13] = second() % 10 + '0';
        Time[9]  = minute() / 10 + '0';
        Time[10] = minute() % 10 + '0';
        Time[6]  = hour()   / 10 + '0';
        Time[7]  = hour()   % 10 + '0';
 
        // update date array
        Date[6]  =  day()   / 10 + '0';
        Date[7]  =  day()   % 10 + '0';
        Date[9]  =  month() / 10 + '0';
        Date[10] =  month() % 10 + '0';
        Date[14] = (year()  / 10) % 10 + '0';
        Date[15] =  year()  % 10 + '0';
      }
  display.setFont(ArialMT_Plain_10);
  display.flipScreenVertically();
  if(gps.location.isValid() || gps.time.isValid() || gps.date.isValid()){
    display.clear();
    display.drawString(0,0, "Location: " + String(gps.location.lat(), 4) +"," + String(gps.location.lng(), 4));
    display.drawString(0,10, String(Date));
    display.drawString(0,20, String(Time));
    //display.drawString(0,30, "Battery: " + String(axp.getBattPercentage()) + "%");
    display.drawString(0,30, "BattVoltage: " + String(axp.getBattVoltage()/1000) + "V");
    display.drawString(0,40, "Satellite: " + String(gps.satellites.value()));
    display.display();
  }`

those are the example code that I use and change according to my need and I'm not sure where is my mistake to begin with, hope you guys can help me to solve this problem, I try to solve this problem for over a week now but still cant get any solution . 

p/s: ignore the coordinate as I'm indoor right now so there will be no coordinate

TD-er, 2022-04-28:
About the code formatting.
You should use 3 backticks at begin and end (on a separate line) and if you like to have syntax highlighting, you can add "c++" right after the first 3 backticks.

About the wrong date, are you sure you have a fix?

Firered97, 2022-04-28:
Sorry bout the format coz its my first time put the code here

Regarding the date, as you can see from the pic i put there, i have try like a few example code but only the time is working for me but not the date, and im not sure whats wrong with the code because i follow exactly from the code, i just changed it a bit becoz im displaying it to oled display, thats all

cyberman54, 2022-04-28:
It looks like your gps has no, this it may have a time but no date information..Gi outdoor and assure you have gps fix.

TD-er, 2022-04-28:
When displaying the time/date, you don't check if the date is valid in your code.
If your GPS module doesn't have a fix, the date can be just about anything.
I have seen it reporting even dates in 2040.

Firered97, 2022-04-28:
how to know if my gps module have a fix? im really confuse

Firered97, 2022-04-28:
do you have any suggestion about how can I get the GPS module to have it fix?

cyberman54, 2022-04-28:
depends on your gps chipset & antenna & if it is a cold or hot start.
If cold start, and weak signal, it can take hours.

Firered97, 2022-04-29:
@cyberman54 is there anyway other from just let the module sit outdoor for hours? because I put mine from yesterday until now so almost 9 hour but still the date wont change

Jens869, 2022-10-17:
if (gps.date.isValid()) is always TRUE !
if (gps.time.isValid()) is always TRUE !

check if month > 0
if (gps.date.day() == 0) { then Date and Time are invalid)