mikalhart/TinyGPSPlus #102: How can I tell why the encoder didn't process my sentences?

RandyPfeifer, 2022-05-06:
Hi, New to this whole Arduino scene.... 
I am running tinyGPSPlus on a Seeeduino XIAO platform with I2C connected to a display and a compass module in addition to the  NEO-6M GPS module.  

when running the example programs everything works just fine so it would seem that I've wired things Ok. 

When I integrate the .h files into my larger program and then try to do the same functions, I almost never get anything to be printed by displayinfo();   I'm using D9, D8 and 9600 baud software serial.  I've ported over all of the setup() and other declarations from the example (kitchensink.ino primarily although I've toyed with all of the examples. 

This snippet of code
while (ss.available() > 0)
  if(gps.encode(ss.read()))  
     displayInfo();

produces no Serial.printf() output. 

I added some code just after the above code to grab the text and print them myself:

delay(2000);  // let the encoder get the first available sentence and print the next one myself. 
 while (ss.available() > 0) {
    char character = ss.read();
      Serial.print(character);

This snippet of code generates output on the serial monitor that looks legit (based on my reading of a couple NMEA sentence definitions I found.   There is apparently something wrong with it but I can't tell what is afoul.   Here's what that output looks like (ignore the x:, y:, z:, etc stuff - it's output from the compass and other calculations in reset of my program).

21:15:44.607 -> $GPRMC,021543.00,A,4335.24758,N,09005.07362,W,0.182,,060522,,,A  x: 722,y: -1513,z: 868,Target: 295.53,Measurement: 295.51,Error: -0.02,Cumulative Error: 6.09
21:15:46.738 -> $GPRMC,021545.00,A,4335.24759,N,09005.07421,W,0.064,,060522,,,A  x: 726,y: -1514,z: 865,Target: 295.53,Measurement: 295.62,Error: 0.08,Cumulative Error: 6.17
21:15:48.550 -> $GPRMC,021547.00,A,4335.24754,N,09005.07421,W,0.238,,060522,,,A  x: 728,y: -1519,z: 865,Target: 295.53,Measurement: 295.61,Error: 0.07,Cumulative Error: 6.24
21:15:50.702 -> $GPRMC,021549.00,A,4335.24748,N,09005.07482,W,0.468,,060522,,,A  x: 720,y: -1514,z: 854,Target: 295.53,Measurement: 295.43,Error: -0.10,Cumulative Error: 6.14
21:15:52.561 -> $GPRMC,021551.00,A,4335.24748,N,09005.07554,W,0.716,,060522,,,A  x: 719,y: -1518,z: 864,Target: 295.53,Measurement: 295.34,Error: -0.19,Cumulative Error: 5.95
21:15:54.662 -> $GPRMC,021553.00,A,4335.24724,N,09005.07558,W,0.529,,060522,,,A  x: 726,y: -1510,z: 868,Target: 295.53,Measurement: 295.68,Error: 0.14,Cumulative Error: 6.10
21:15:56.595 -> $GPRMC,021555.00,A,4335.24727,N,09005.07596,W,0.548,,060522,,,A  x: 728,y: -1513,z: 867,Target: 295.53,Measurement: 295.70,Error: 0.16,Cumulative Error: 6.26

So my question is, is there a good way to tell why the encoder didn't successfully encode the above sentences and allow the displayInfo() routine to successfully print anything?

Any help/suggestions would be much appreciated.  I don't doubt that the encoder found something reprehensible but I can't figure out what it is.  I really don't want to try to parse these sentences myself. 

Thanks in advance.

RandyPfeifer, 2022-05-06:
Oh but when I added a processed character counter it wasn't incrementing very fast.  after a few other changes it is now incrementing in bursts.  Checksum errors are also occurring on occasion (but not as frequently as the character count is increasing.   

I'll keep poking at it.  

Thanks again for your guidance.

cyberman54, 2022-05-06:
Your display routine displayInfo() prints only data, if gps got a fix. But it seems it doesn't have one: "sentences with no fix=0". Thus, nothing is printed.

Bursts are normal with NMEA protocol, since gps data is coming in cyclic.

RandyPfeifer, 2022-05-06:
"Sentences with fix=0" yet there are clearly sentences with date, time and position data in them that are passing the checksum and are incrementing the processed character count (albeit in bursts)....

RandyPfeifer, 2022-05-06:
Confirm something for me... If I read this code right displayInfo() will be called for every character received from the GPS module.  Or does gps.encode only return true if it completes a sentence?

while (ss.available() > 0)
if(gps.encode(ss.read()))
displayInfo();

RandyPfeifer, 2022-05-06:
NVM it looks like displayInfo() is invoked only when a sentence is encoded (not every character)

cyberman54, 2022-05-06:
encode returns true, after a valid sentence was processed, not every character.

RandyPfeifer, 2022-05-06:
After looking some more I think I have more fundamental problems.  while the example Tinygpsplus_sample.ino runs and reports good results, when I print out characters processed, failed checksums and sentences with a fix, I get a lot of errors:
[image]

The number of checksum errors are close to the number of sentences with a fix.  Both numbers increase about once per second and every so often the Fix count goes up by one and the checksum fail count doesn't.... So now my question (for me to figure out) is why doesn't even the example code run well....

cyberman54, 2022-05-06:
Check your timing, wiring, serial interface, and configuration of gps chip. This won't be an issue of Tinygps, it usually runs smooth and stable.

RandyPfeifer, 2022-05-06:
Very good.   Thanks again for your attention and support. 

much appreciated.

mikalhart, 2022-09-04:
All good here, @RandyPfeifer?  Good rule of thumb is don't put any delay()s in your processing loop and do as little Serial.print...ing as possible.