mikalhart/TinyGPSPlus #103: Support for Raspberry Pi Pico RP2040

zpm1066, 2022-08-30:
Thanks for the fantastic work on TinyGPSPlus.

Do you have plans to portTinyGPSPlus to the Raspberry Pi Pico RP2040? 

With the RP2040 gathering a lot of support over 4th last year, I and quite a few other folks would certainly like to see TinyGPSPlus support the Pico. Please consider. Thanks!

cyberman54, 2022-08-30:
TinyGPSplus is a NMEA parser, written in C++, thus it's platform independent.So there is no need to port anything to a certain platform. Just use it as is in you C++ toolchain for your platform.

zpm1066, 2022-09-02:
@cyberman54, Yes, of course! I have both u-blox Neo 6MN and 8MN working with Pico W. Thanks.

mikalhart, 2022-09-04:
Great!  @zpm1066, did you have to modify anything to get it to work on the Pi?  (I worry about possible Arduino dependencies like the 'millis' function.  Haven't played with the Pi yet to understand its working environment.)

zpm1066, 2022-09-04:
Hi @mikalhart, no. I used your TinyGPSPlus as is. Used Serial1 UART on the Raspberry Pico W.
Thank you for the great NMEA parser.

closed 2022-09-04 by zpm1066