mikalhart/TinyGPSPlus #107: date.isValid and time.isValid are always true

Jens869, 2022-10-17:
I'm currently working on a watch that uses GPS to tell the time.
But the error is also in the example DeviceExample.ino

`Serial.print(F("  Date/Time: "));
  if (gps.date.isValid())
  {
    Serial.print(gps.date.month());
    Serial.print(F("/"));
    Serial.print(gps.date.day());
    Serial.print(F("/"));
    Serial.print(gps.date.year());
  }else
  {
    Serial.print(F("INVALID"));
  }

 if (gps.date.isValid()) {
 // do something 
} else
  {
    Serial.print(F("INVALID"));
  }

`

 if (gps.date.isValid()) is always TRUE, even if GPS fails.
gps.date.month() and gps.date.day() are 0. 
Valid values ​​are month 1 to 12 - day 1 to 31.

if (gps.time.isValid()) is always TRUE, even if GPS fails.

Work around: check if the month or day is 0 Zerro

Jens869, 2022-10-17:
Serial Monitor

reset
Location: INVALID  Date/Time: INVALID INVALID gps.date.isValid() =0 gps.date.age() =4294967295
Location: INVALID  Date/Time: 0-0-2000 00:00:00   gps.date.isValid()=1   gps.date.age()=78
Location: INVALID  Date/Time: 0-0-2000 00:00:00.  gps.date.isValid()=1   gps.date.age()=125
Location: INVALID  Date/Time: 0-0-2000 00:00:00.  gps.date.isValid()=1   gps.date.age()=170

mikalhart, 2022-10-17:
Hi @Jens869,

First of all, may I ask what watch you are playing with?  We are enjoying working with the LilyGo T-Watch 2020 V2, which also has GPS.  https://www.aliexpress.com/item/1005002264354524.html.  Great fun!

It was arguably a weak naming choice, but isValid() == true means only "**at least one properly formatted sentence containing this field has been received**" and not "**this field was populated with data that has been verified to be correct**".  I'm guessing that your GPS has sent a sentence which contains a blank date/time field--which is perfectly **Valid**, if not immediately useful.

Early on we decided that, in the interest of staying **Tiny**, the library would act strictly as a parser, and not a validator.  Your sort of workaround is exactly what is needed.  For example, one of the devices we sometimes deploy sends "January 1, 2080" as the date until it gets a fix.  We have to check for this condition outside TinyGPS++.

Thanks for the comments.