mikalhart/TinyGPSPlus #114: Issues with extracting the fix data using custom NMEA sentence extraction

classbproject, 2023-02-03:
I'm trying to extract the GPS Fix data using the custom NMEA sentence extraction but it does not seem to work. 

The GPS fix data is present in the GPGGA sentence as the 6th element. In the code, I have,

```
TinyGPSPlus gps;
TinyGPSCustom fixStatus(gps, "GPGGA", 6);
```

and in the loop() function, I have,

`Serial.print("Fix status: "); Serial.println(fixStatus.value());`

However, I do not see `1` though the GPGGA sentence does show it. 

The full code is [here](https://pastebin.com/whxm3fcJ).

cyberman54, 2023-02-03:
Using GNGGA means you want GLONASS navigation.
Are you using a current version of TinyGPS?
GLONASS support was added later: https://github.com/mikalhart/TinyGPSPlus/commit/ba91ede894eb6491b1f2444cbf0b3f8309747bfb

classbproject, 2023-02-05:
I never realised that I was using an older version. I added the library from the Arduino IDE and assumed it would be the latest version. I've now updated to the latest release. I now see the fix value as `-68.1`. Not sure how to interpret this value.

adamgarbo, 2023-07-19:
@classbproject I use the following code with good success to determine the fix and validity of my received data:

```
TinyGPSPlus gnss;

// Custom TinyGPS objects to store fix and validity information
TinyGPSCustom gnssFix(gnss, "GNGGA", 6); // Fix quality
TinyGPSCustom gnssValidity(gnss, "GNRMC", 2); // Validity
```

And then to check if the sentences are valid, I check the custom fix and validity values, as well as the number of satellites:
```
// Check if NMEA sentences have a valid fix and are not stale
if ((gnssFix.value() > 0 && gnssFix.age() < 1000) &&
    (String(gnssValidity.value()) == "A" && gnssValidity.age() < 1000) &&
    gnss.satellites.value() > 0)
{
```

Cheers,
Adam

cyberman54, 2023-07-23:
> // Check if NMEA sentences have a valid fix and are not stale
> if ((gnssFix.value() > 0 && gnssFix.age() < 1000) &&
>     (String(gnssValidity.value()) == "A" && gnssValidity.age() < 1000) &&
>     gnss.satellites.value() > 0)
> {

This doesn't make sense to me, since it adds overhead to calculate a status which is already processed by TinyGPS. Simply check if `location.isValid()` and/or `altitude.isValid()`. TinyGPS commits these values, if a fix was present when these values arrived.

To detect a fix TinyGPS does exactly the same as your code suggests:

```
case COMBINE(GPS_SENTENCE_GPRMC, 2): // GPRMC validity
      sentenceHasFix = term[0] == 'A';

case COMBINE(GPS_SENTENCE_GPGGA, 6): // Fix data (GPGGA)
      sentenceHasFix = term[0] > '0';
```

adamgarbo, 2023-07-23:
Thanks for your input.

To give you some background, I previously had an issue with the validity check I was using where I would lose contact with my instruments for a month.

```
if ((gps.location.isValid() && gps.date.isValid() && gps.time.isValid()) &&
    (gps.location.isUpdated() && gps.date.isUpdated() && gps.time.isUpdated())) {}
```

Somehow, data with incorrect dates were making it past the check and causing havoc with my real-time clock alarms. I spoke to Mikal directly about this and it turns out `date.isValid()` and `time.isValid()` are not very rigorous and can easily report an incorrect date and/or time as valid.

After that, I started doing things explicitly myself to ensure the validity check worked. I don't mind the redundancy.

Cheers,
Adam

cyberman54, 2023-07-23:
isValid() means the data was seen by TinyGPS in a sentence with fix. It does not mean the sentence is not stale. Thus, you should combine this with checking the age of the sentence. That's all. See my example.

adamgarbo, 2023-07-23:
Thanks for your feedback. Glad to have confirmation my code above is the correct (albeit redundant) approach.

classbproject, 2023-12-27:
> @classbproject I use the following code with good success to determine the fix and validity of my received data:
>
> ```
> TinyGPSPlus gnss;
>
> // Custom TinyGPS objects to store fix and validity [...]

I was able to get a fix check with the following code:

```
bool checkGPSFix()
{
  while ((gpsSerial.available() > 0))
  {
    if (gps.encode(gpsSerial.read()))
    {
      if ((gps.location.isValid() && gps.date.isValid() && gps.time.isValid()) &&
          (gps.location.isUpdated() && gps.date.isUpdated() && gps.time.isUpdated()) &&
          (gps.satellites.value() > 0))
      {
        fixStatus = true;
        Serial.print("Fix status: ");
        Serial.println(fixStatus);
        return fixStatus;
      }
    }
  }
  return 0;
}
```
My apologies for the delayed response. I got a chance to work on this again now. I saw your [comment](https://github.com/mikalhart/TinyGPSPlus/issues/70#issuecomment-922287142) on another thread for the fix check. For some reason the custom TinyGPS function won't work. I need to do more experiments to see why that won't work. For now this works though. Thanks!