mikalhart/TinyGPSPlus #115: No Altitude and Satellites values.

ad1k4h, 2023-03-03:
I have this function. This needs to be populate the values to global function:

void startGPS()
{
  if (gps.location.isValid())
  {
    
    latitude = String(gps.location.lat(), 10);
    longitude = String(gps.location.lng(), 10);
    altitude = String(gps.altitude.meters(), 2); // Altitude in meters (double)
    speed = String(gps.speed.kmph(), 2);
    satellites = String(gps.satellites.value());
    Serial.println(satellites);
    GPS_Status(1);
  }
  else
  {
    Serial.print(F("Searching for Satellites..."));
    GPS_Status(0);
  }

}

I getting latitude, longitude and speed, but altitude and satellites are empty. Am i trying it good?

mikalhart, 2023-03-03:
Altitude and satellites come from the $GPGGA sentence, while the others are available in $GPRMC.  Best guess?  Either your device has been configured to disable $GPGGA sentences, or you are calling startGPS() before the first $GPGGA sentence has been retrieved.

 

Are you calling startGPS repeatedly?  If so, you should try temporarily removing the Serial.print(F("Searching for Satellites...")); line, because it may be introducing delays that cause serial overflow.

 

Last thing to try… change

if (gps.location.isValid())

to

 

if (gps.location.isValid() && gps.satellites.isValid())

 

Good luck!

 

M

 

I getting latitude, longitude and speed, but altitude and satellites are empty. Am i trying it good or its a bug?

ad1k4h, 2023-03-03:
Changing to this:

if (gps.location.isValid() && gps.satellites.isValid())

**Solved the issue.** Thank you so much, and sorry for the issue. May thanks! <3

closed 2023-03-03 by ad1k4h

mikalhart, 2023-03-03:
> Changing to this:
> 
> if (gps.location.isValid() && gps.satellites.isValid())
> 
> **Solved the issue.** Thank you so much, and sorry for the issue. May thanks! <3

Excellent!