mikalhart/TinyGPSPlus #119: What a great gps library

bazjay, 2023-04-17:
Please advise the licence that the code is distributed under.

adamgarbo, 2023-07-19:
@bazjay if you look at the code, you will see it states:

```
This library is free software; you can redistribute it and/or
modify it under the terms of the GNU Lesser General Public
License as published by the Free Software Foundation; either
version 2.1 of the License, or (at your option) any later version.
```

@mikalhart it could be helpful to add a LICENSE file to your repository.

scuba-hacker, 2023-07-19:
D'oh! How on earth did I miss that @adamgarbo ... !

IgnoreWarnings, 2023-08-20:
> @bazjay if you look at the code, you will see it states:
>
> ```
> This library is free software; you can redistribute it and/or
> modify it under the terms of the GNU Lesser General Public
> License as [...]

I recommend adding a License file, since it will show directly in github and spare users from looking up this thread.