mikalhart/TinyGPSPlus #127: Location: INVALID

Nikitanagar, 2023-10-26:
Hello, I am using gps nut it shows error. i am sharing the code please, let me know the solution.

#define TINY_GSM_MODEM_SIM7600  // SIM7600 AT instruction is compatible with A7670
#define SerialAT Serial1
#define SerialMon Serial
#define TINY_GSM_USE_GPRS true
#include <TinyGsmClient.h>
//#include <WiFi.h>
#include <Firebase_ESP_Client.h>
#include <Wire.h>
#include <L3G.h>
#include <TinyGPSPlus.h>
#define TINY_GSM_DEBUG SerialMon
#define TINY_GSM_USE_WIFI false
#include <addons/TokenHelper.h>
#include <addons/RTDBHelper.h>

TinyGPSPlus gps;

// See all AT commands, if wanted
 #ifdef DUMP_AT_COMMANDS
#include <StreamDebugger.h>
StreamDebugger debugger(SerialAT, SerialMon);
TinyGsm        modem(debugger);
#else
TinyGsm        modem(SerialAT);
#endif
TinyGsmClient client(modem);

// Define the serial console for debug prints, if needed

// set GSM PIN, if any
#define GSM_PIN ""

// Your GPRS credentials, if any
const char apn[] = "";
const char gprsUser[] = "";
const char gprsPass[] = "";

#define uS_TO_S_FACTOR 1000000ULL // Conversion factor for micro seconds to seconds
#define TIME_TO_SLEEP 600         // Time ESP32 will go to sleep (in seconds)

#define UART_BAUD 115200

//#define PIN_TX 26
//#define PIN_RX 27

static const int RXPin = 16, TXPin = 17;
static const uint32_t GPSBaud = 9600;
int sensor = 34; 
int GX;
int GY;
int GZ;
L3G gyro; 

TinyGsmClient gsm_client(modem);

float MQ3;
int Unique_ID;
double Latitude;
double Longitude;

#define API_KEY "......."
#define DATABASE_URL "......" 
#define USER_EMAIL "......."
#define USER_PASSWORD "......."

FirebaseData fbdo;
FirebaseAuth auth;
FirebaseConfig config;

unsigned long sendDataPrevMillis = 0;
unsigned long count = 0;
bool signupOK = false;

void setup()
{

  Serial.begin(9600);
  Serial2.begin(9600);
  delay(3000);

    Wire.begin();
    delay(10);
    DBG("Wait...");

    //SerialAT.begin(UART_BAUD, SERIAL_8N1, PIN_RX, PIN_TX);

    DBG("Initializing modem...");
    if (!modem.init())
    {
        DBG("Failed to restart modem, delaying 10s and retrying");
        return;
    }

    /*
    2 Automatic
    13 GSM Only
    14 WCDMA Only
    38 LTE Only
    */
    modem.setNetworkMode(38);
    if (modem.waitResponse(10000L) != 1)
    {
        DBG("setNetworkMode faill");
    }

    String name = modem.getModemName();
    DBG("Modem Name:", name);

    String modemInfo = modem.getModemInfo();
    DBG("Modem Info:", modemInfo);

    Serial_Printf("Firebase Client v%s\n\n", FIREBASE_CLIENT_VERSION);

    /* Assign the api key (required) */
    config.api_key = API_KEY;

    /* Assign the user sign in credentials */
    auth.user.email = USER_EMAIL;
    auth.user.password = USER_PASSWORD;

    config.database_url = DATABASE_URL;
    config.token_status_callback = tokenStatusCallback; // see addons/TokenHelper.h
    fbdo.setGSMClient(&gsm_client, &modem, GSM_PIN, apn, gprsUser, gprsPass);

   
    Firebase.reconnectNetwork(true);
    fbdo.setBSSLBufferSize(4096 /* Rx buffer size in bytes from 512 - 16384 */, 1024 /* Tx buffer size in bytes from 512 - 16384 */);
    Firebase.setDoubleDigits(5);
    Firebase.begin(&config, &auth);
     if (!gyro.init())
  {
  Serial.println("Failed to autodetect gyro type!");
  while (1);
 
}
  gyro.enableDefault();
}

void loop()

 {
   if (millis() > 5000 && gps.charsProcessed() < 10)
  {
    Serial.println(F("No GPS detected: check wiring."));
  }

   display();

    // Firebase.ready() should be called repeatedly to handle authentication tasks.
  if (Firebase.ready() && (millis() - sendDataPrevMillis > 15000 || sendDataPrevMillis == 0))
    {
        sendDataPrevMillis = millis();
   
    if (Firebase.RTDB.setInt(&fbdo, "/GSM/Unique_ID", 001)) {
      Serial.println("Unique_ID Inserted");
    }
    
    if (Firebase.RTDB.setInt(&fbdo, "/GSM/Latitude", Latitude)) {
      Serial.println("Latitude Inserted");
    }
    else {
      Serial.println(fbdo.errorReason());
    }
    
    
    if (Firebase.RTDB.setInt(&fbdo, "/GSM/Longitude", Longitude)) {
      Serial.println("Longitude Inserted");
    }
    else {
      Serial.println("Longitude Value Not Inserted");
      Serial.println(fbdo.errorReason());
    }
    
    
    if (Firebase.RTDB.setInt(&fbdo, "/GSM/MQ3", MQ3)) {
      Serial.println("MQ3 Value Inserted");
    }
    else {
      Serial.println("MQ3 Value Not Inserted");
      Serial.println(fbdo.errorReason());
    }
 
    
    if (Firebase.RTDB.setInt(&fbdo, "/GSM/Gx", GX)) {
      Serial.println("Gx Value Inserted");
    }
    else {
      Serial.println("Gx Value Not Inserted");
      Serial.println(fbdo.errorReason());
    }
  
    if (Firebase.RTDB.setInt(&fbdo, "/GSM/Gy", GY)) {
      Serial.println("Gy Value Inserted");
    }
    else {
      Serial.println("Gy Value Not Inserted");
      Serial.println(fbdo.errorReason());
    }
    

    if (Firebase.RTDB.setInt(&fbdo, "/GSM/Gz", GZ)) {
      Serial.println("Gz Value Inserted");
    }
    else {
      Serial.println("Gz Value Not Inserted");
      Serial.println(fbdo.errorReason());
    }

  }
  else{
    Serial.println("outside the loop");
  }

  display();
}

 void display(){
     gps.encode(Serial2.read());
   
    if (gps.location.isValid()) {
    
		  Latitude = (gps.location.lat(),6);
      Longitude = (gps.location.lng(),6);
      Serial.print(Latitude,6);
      Serial.print(" , ");
      Serial.println(Longitude,6);
	
	  } else {
		  Serial.print(F("INVALID"));
	  }
      

    MQ3 = analogRead(sensor); 
    Serial.print("MQ3: ");
    Serial.println(MQ3);

    gyro.read();
  Serial.println("G ");
  GX = (gyro.g.x);
  Serial.println(GX);
  GY = (gyro.g.y);
  Serial.println(GY);
  GZ = (gyro.g.z);
  Serial.println(GZ);

 }
Please let me know the error.

JasonPittenger, 2023-10-27:
I see you define the baud for GPS, but that is never used again.

static const uint32_t GPSBaud = 9600;

That's fine, because you later initialize two serial ports at 9600 baud.
Serial.begin(9600);
Serial2.begin(9600);

It looks like Serial is your debugging port, based on usage in your code.
I assume then that Serial2 is your GPS serial port.
I only see you reading it inside the function display(), which is called in loop()

void display()
{
gps.encode(Serial2.read());
}

This will not work.
The read() function returns the first byte of incoming serial data available (or -1 if no data is available).
This means you're only grabbing a single byte and trying to display the data.
Instead, put this inside loop()
```
uint16_t GPSbytesAvailable;
GPSbytesAvailable = Serial2.available();
if(GPSbytesAvailable)
{
	while(GPSbytesAvailable--)
	{
		gps.encode(Serial2.read());
	}
}
```
or
```
while (Serial2.available() > 0)
{
	gps.encode(Serial2.read);
}
```

The first code is a little faster because it doesn't repeatedly call Serial2.available();

svdrummer, 2023-10-28:
I was wondering why "GPSbytesAvailable--" is faster?

TD-er, 2023-10-28:
Nope, NOT calling serial.available() is faster as you need to query something that's for sure a call to some hardware.
And it might be possible you keep on running longer when checking again for new available bytes, where in the first code example you only check this once and just assume no more bytes have entered the serial buffer.