mikalhart/TinyGPSPlus #130: Pull requests.

svdrummer, 2023-11-09:
Are these going to be linked into the mail code. There are 11 already with more on the side.

mikalhart, 2023-11-11:
Hi!

@svdrummer Could you explain that first sentence a bit?  I've been looking at the pull requests here: many are interesting.

General request for comment: I'm thinking about making more extensive version of this library for more powerful micros (e.g. ESP32) that would
a) support all the various satellite systems
b) retain the latest copy of each sentence per each keyword (e.g. GPGGA)
c) run asynchonously in a different thread, so you wouldn't have to worry about always manually and frequently calling encode() 

Thoughts?

svdrummer, 2023-11-11:
Judging by the groups I follow and the trends in what users are using, it would seem no one uses arduino..rephrase..even arduino is dropping the old and now uses an ESP32 using the arduino foot print. Most of the users of your library seem to use ESP8266 or ESP32. As the price of 8266 and 32 are so close now, and you get more bang per buck in an ESP32, I only use ESP32s. 
A) As modern GPS's support more satellite systems, we have been adding this manually as advised by other branches of your library.
B) I am not sure, as I only want to know where I am in the most accurate manner.
C) I like this concept of using another core for the library, albeit, I have seen this done using your library on you tube. I don't like to drift away from your library, as Yours works and has been tested by time..It is a sudo standard when using a GPS module.
I think controlling drones would need the speed, and leave core ) for the main code.

mikalhart, 2023-11-11:
Thanks for the feedback, @svdrummer.  I, too, almost exclusively use ESP32 boards these days.