mikalhart/TinyGPSPlus #135: No Data coming from module

dani9084, 2024-02-01:
Hello, I am facing an interesting problem using a neo 6m gps module and 2 ESP32 boards. 
I am trying to run this example code: 
`/*
 * This ESP32 code is created by esp32io.com
 *
 * This ESP32 code is released in the public domain
 *
 * For more detail (instruction and wiring diagram), visit https://esp32io.com/tutorials/esp32-gps
 */

#include <TinyGPS++.h>

#define GPS_BAUDRATE 9600  // The default baudrate of NEO-6M is 9600

TinyGPSPlus gps;  // the TinyGPS++ object

void setup() {
  Serial.begin(9600);
  Serial2.begin(GPS_BAUDRATE);

  Serial.println(F("ESP32 - GPS module"));
}

void loop() {
  if (Serial2.available() > 0) {
    if (gps.encode(Serial2.read())) {
      if (gps.location.isValid()) {
        Serial.print(F("- latitude: "));
        Serial.println(gps.location.lat());

        Serial.print(F("- longitude: "));
        Serial.println(gps.location.lng());

        Serial.print(F("- altitude: "));
        if (gps.altitude.isValid())
          Serial.println(gps.altitude.meters());
        else
          Serial.println(F("INVALID"));
      } else {
        Serial.println(F("- location: INVALID"));
      }

      Serial.print(F("- speed: "));
      if (gps.speed.isValid()) {
        Serial.print(gps.speed.kmph());
        Serial.println(F(" km/h"));
      } else {
        Serial.println(F("INVALID"));
      }

      Serial.print(F("- GPS date&time: "));
      if (gps.date.isValid() && gps.time.isValid()) {
        Serial.print(gps.date.year());
        Serial.print(F("-"));
        Serial.print(gps.date.month());
        Serial.print(F("-"));
        Serial.print(gps.date.day());
        Serial.print(F(" "));
        Serial.print(gps.time.hour());
        Serial.print(F(":"));
        Serial.print(gps.time.minute());
        Serial.print(F(":"));
        Serial.println(gps.time.second());
      } else {
        Serial.println(F("INVALID"));
      }

      Serial.println();
    }
  }

  if (millis() > 5000 && gps.charsProcessed() < 10)
    Serial.println(F("No GPS data received: check wiring"));
}
`

I've tried the code on 2 different ESP32 boards, tried switching TX and RX cables on both, but nothing seems to work. I only get INVALID in the serial port.

TD-er, 2024-02-01:
Please use 3 backticks at begin and end of the code, each on a single line.
The single backtick is for inline code `like this`.

Also which pins do you use for Serial2?

dani9084, 2024-02-01:
I'm using pin 16 and 17. 
According to the pinout its RX2 and TX2.

TD-er, 2024-02-01:
For ESP32 you better give the pins you use to the begin argument or call `setPins` before calling `Serial2.begin()`

Here the function signature for `setPins`
```c++
    // Negative Pin Number will keep it unmodified, thus this function can set individual pins
    // setPins() can be called after or before begin()
    // When pins are changed, it will detach the previous ones
    bool setPins(int8_t rxPin, int8_t txPin, int8_t ctsPin = -1, int8_t rtsPin = -1);
```

Here for `begin`
```c++
    // When pins are changed, it will detach the previous ones
    // if pin is negative, it won't be set/changed and will be kept as is
    // timeout_ms is used in baudrate detection (ESP32, ESP32S2 only)
    // invert will invert RX/TX polarity
    // rxfifo_full_thrhd if the UART Flow Control Threshold in the UART FIFO (max 127)
    void begin(unsigned long baud, uint32_t config=SERIAL_8N1, int8_t rxPin=-1, int8_t txPin=-1, bool invert=false, unsigned long timeout_ms = 20000UL, uint8_t rxfifo_full_thrhd = 112);
```
So you can pick either one you like best.

dani9084, 2024-02-01:
Solved it. Apparently it wasn't getting a clear enough signal, even though the LED was blinking on the board. I put the module outside and I'm getting readings from the module. Thank you for the help and sorry for waisting your time.

closed 2024-02-01 by dani9084