mikalhart/TinyGPSPlus #136: GPS TEL0094

PoloLastik, 2024-02-04:
## Struggle with TinyGPSPlus and GPS TEL0094
Hello,

I am a beginner on Arduino. I am using a GPS TEL0094 pluged on a Adafruit Feather 32u4 RFM95 LoRa Radio. I have my GPS Module which works with the lights (Power light red constantly and green lignht (PPS light) which alternate between On and off every second). My pineout is the following 

GPS -> Feather32u4
VCC -> 3.3 V
GND -> GND 
RX -> Pin 6
TX -> Pin 5
EN -> Nothing
PPS -> Nothing

Here is my Code
'''cpp – C++
#include <SoftwareSerial.h>
 
#include <TinyGPS.h>
 
/* This sample code demonstrates the normal use of a TinyGPS object.
   It requires the use of SoftwareSerial, and assumes that you have a
   4800-baud serial GPS device hooked up on pins 4(rx) and 3(tx).
*/
 
TinyGPS gps;
SoftwareSerial ss(6, 5);
 
void setup()
{
  Serial.begin(115200);
  ss.begin(4800);
  Serial.print("Simple TinyGPS library v. "); Serial.println(TinyGPS::library_version());
  Serial.println("by Mikal Hart");
  Serial.println();
}
 
void loop()
{
  bool newData = false;
  unsigned long chars;
  unsigned short sentences, failed;
 
  // For one second we parse GPS data and report some key values
  for (unsigned long start = millis(); millis() - start < 1000;)
  {
    while (ss.available())
    {
      char c = ss.read();
      // Serial.write(c); // uncomment this line if you want to see the GPS data flowing
      if (gps.encode(c)) // Did a new valid sentence come in?
        newData = true;
    }
  }
 
  if (newData)
  {
    float flat, flon;
    unsigned long age;
    gps.f_get_position(&flat, &flon, &age);
    Serial.print("LAT=");
    Serial.print(flat == TinyGPS::GPS_INVALID_F_ANGLE ? 0.0 : flat, 6);
    Serial.print(" LON=");
    Serial.print(flon == TinyGPS::GPS_INVALID_F_ANGLE ? 0.0 : flon, 6);
    Serial.print(" SAT=");
    Serial.print(gps.satellites() == TinyGPS::GPS_INVALID_SATELLITES ? 0 : gps.satellites());
    Serial.print(" PREC=");
    Serial.print(gps.hdop() == TinyGPS::GPS_INVALID_HDOP ? 0 : gps.hdop());
  }
  gps.stats(&chars, &sentences, &failed);
  Serial.print(" CHARS=");
  Serial.print(chars);
  Serial.print(" SENTENCES=");
  Serial.print(sentences);
  Serial.print(" CSUM ERR=");
  Serial.println(failed);
  if (chars == 0)
    Serial.println("** No characters received from GPS: check wiring **");
}
'''
Here is my output : 
[image]

I am aware that the Rx and Tx must be inversed and I took care of that. I know that the GPS needs 30 seconds to start. I am not quite sure of the PPS and EN purpose despite I read the GPS datasheet several times. 

Could somoene help me with this issue ?

svdrummer, 2024-02-04:
We all started somewhere...However... these githubs are library issues, not really for hardware or script debugging.
Check your GPS unit manual, and you will most likely find that the GPS is 9600 baud and not 4800 baud. If which case, 
change  ss.begin(4800); 
to ss.begin(9600);

A quick google on MY part shows the following for this module.
UBX-G7020-KT solution
9600bps (default) (4800, 9600, 19200, 38400, 57600, 115200, 230400, 460800, 921600) TTL interface
5V at 30mA (support 3.3V to 5V)
56-channel receiver
Extremely high sensitivity -161dBm

Notice the default 9600bps(default) line.

Hope this helps. Google is your friend.

mikalhart, 2024-02-04:
@PoloLastik, I note that you have declared pin 6 as your RX pin in the SoftwareSerial ss declaration, but then written that you connected pin 6 to the GPS RX pin.  Remember that RX from the microcontroller's viewpoint is TX from the GPS module's.  The easiest fix at this point would be to change your software a bit:

```cpp – C++
SoftwareSerial ss(5, 6);
```

And yes, @TD-er is right: most modules these days are 9600 baud.

TD-er, 2024-02-04:
This time it wasn't me who replied :)

mikalhart, 2024-02-04:
Ha, whoops, indeed.  Credit where credit is due: sorry, @svdrummer :)