mikalhart/TinyGPSPlus #137: ATGM336H Not Encoding

WaynePauly, 2024-03-19:
We switched from the NEO-6M to the ATGM336H in our balloon trackers and the software won't encode a fix.
I tried the Device Example with no results either, but the raw NMEA sentences look normal to me and indicate a fix.
Any known issues using the 336H with TinyGPSPlus or suggestions? Which sentence does the library try to parse?
Wayne

mikalhart, 2024-03-19:
Hi Wayne—

 

TinyGPSPlus looks at G*GGA and G*RMC sentences by default.  Can you possible email some sample NMEA?  (sundial @ sundial.com) I bet I can figure it out and make fix if needed.

 

Interested to hear about your balloon tracker!  We are flying a balloon ourselves for the 8 April eclipse.

 

Mikal

WaynePauly, 2024-03-23:
The ATGM336H has many attractive qualities, but playing well with TinyGPSPlus is not one of them. With Mikal's help we determined it decoded the $GNGGA fine but always reported invalid. I added a custom TinyGPSCustom ATGMfix(gps, "GNGGA", 6); // $GNGGA sentence, 6th element to pluck out the fix quality and decided if it was valid. Works fine and we are ready to launch balloons again. Issue closed.

guming56, 2024-04-04:
I also use this GPS module, can you explain it in more detail? Mine cannot be parsed using the hardware serial port, but the software serial port can parse NMEA.

WaynePauly, 2024-04-04:
I have never had much luck using TinyGPSplus with anything other than true
Arduino's.
I could never get it to output results with the 336 chip using TinyGPSplus but
could get it to parse using the new custom field commands feature.

Try asking ChatGPS to "code a NMEA parser for the ATGM336H GPS module". It
produced marvelous code that worked perfectly, but you do have to convert the
fields from text to useful numbers.

Good luck

Received: Thu, 04 Apr 2024 07:48:01 AM EDT

https://github.com/mikalhart/TinyGPSPlus/issues/137#issuecomment-2036951567