mikalhart/TinyGPSPlus #149: Compatibility with Arduino Mega 2560 R3

Ooip83, 2025-07-28:
Hi, I am using GT-U8 gps module from [https://shopee.com.my/product/8822702/8216753731?d_id=6f320&uls_trackid=5371clh6011a&utm_content=X8dqHZFAEE1j7L9j7tApAEN8niT] (url) to interface with Arduino Mega 2560 R3. It needs a logic level converter and I am using TXS0108E 8-Bit Bi-Directional Logic Level Converter.

I am using pin 19 for RX1 and 18 for TX1, however it never works, it stops at waiting for GPS signal. Below is my code:
```
#include <TinyGPSPlus.h>

// TinyGPS++ instance
TinyGPSPlus gps;

void setup() {
  Serial.begin(115200);         // For Serial Monitor
  Serial1.begin(9600);          // GPS on Serial1 (RX1 = pin 19, TX1 = pin 18)

  Serial.println(F("GT-U8 GPS with TinyGPS++ on Arduino Mega"));
  Serial.print(F("TinyGPS++ library version: "));
  Serial.println(TinyGPSPlus::libraryVersion());
  Serial.println(F("Waiting for GPS signal..."));
}

void loop() {
  // Feed GPS data
  while (Serial1.available()) {
    gps.encode(Serial1.read());
  }

  // Only print if there's a valid fix
  if (gps.location.isUpdated()) {
    Serial.print("Latitude: ");
    Serial.println(gps.location.lat(), 6);

    Serial.print("Longitude: ");
    Serial.println(gps.location.lng(), 6);

    Serial.print("Altitude (m): ");
    Serial.println(gps.altitude.meters());

    Serial.print("Speed (km/h): ");
    Serial.println(gps.speed.kmph());

    Serial.print("Satellites: ");
    Serial.println(gps.satellites.value());

    Serial.print("Date: ");
    if (gps.date.isValid()) {
      Serial.print(gps.date.day());
      Serial.print("/");
      Serial.print(gps.date.month());
      Serial.print("/");
      Serial.println(gps.date.year());
    } else {
      Serial.println("Invalid");
    }

    Serial.print("Time (UTC): ");
    if (gps.time.isValid()) {
      Serial.print(gps.time.hour());
      Serial.print(":");
      Serial.print(gps.time.minute());
      Serial.print(":");
      Serial.println(gps.time.second());
    } else {
      Serial.println("Invalid");
    }

    Serial.println(F("--------------------------"));
  }

  delay(1000);  // Adjust as needed
}
```

The LED on the module is blinking. I tested it on 3.3V logic device such as ESP32, and it works. It looks like the problem is on Arduino Mega 2560 R3. Anyone encounter this issue before and is there any fix? Thanks

svdrummer, 2025-07-29:
put a print to serial statement in here

while (Serial1.available()) {
    gps.encode(Serial1.read());
  }

tens on thousands use this library. the error must be in your hardware
setup. Are you getting any data from gps module etc

Ooip83, 2025-07-29:
@svdrummer I have tried, no luck, i put a Serial.println("Working"); soon after the gps.encode(Serial1.read()); but no luck in getting valid values in the following code.

Now i think there's a problem in how to use UART port for Arduino Mega 2560 R3, i am currently using this code for ESP32 and it works fine

```
#include <TinyGPS++.h>

// Create TinyGPS++ object
TinyGPSPlus gps;

// Use Serial2 for hardware UART on GPIO16 (RX2), GPIO17 (TX2)
#define RXD2 16
#define TXD2 17

void setup() {
  // Begin Serial Monitor
  Serial.begin(115200);
  while (!Serial);

  // Begin GPS Serial on UART2
  Serial2.begin(9600, SERIAL_8N1, RXD2, TXD2);

  Serial.println(F("GT-U8 GPS with TinyGPS++ on ESP32 (Hardware Serial)"));
}

void loop() {
  // Read incoming GPS data
  while (Serial2.available() > 0) {
    char c = Serial2.read();
    gps.encode(c);
  }

  // Display info if valid
  if (gps.location.isUpdated()) {
    Serial.print(F("Lat: ")); Serial.println(gps.location.lat(), 6);
    Serial.print(F("Lng: ")); Serial.println(gps.location.lng(), 6);
    Serial.print(F("Satellites: ")); Serial.println(gps.satellites.value());
    Serial.print(F("Altitude: ")); Serial.println(gps.altitude.meters());
    Serial.print(F("HDOP: ")); Serial.println(gps.hdop.hdop());
    Serial.print(F("Speed: ")); Serial.println(gps.speed.kmph());
    Serial.println("----------------------");
  }

  // Optional: timeout message
  static unsigned long lastCheck = 0;
  if (millis() - lastCheck > 5000) {
    if (!gps.location.isUpdated()) {
      Serial.println(F("No GPS data received: check wiring, power, and signal"));
    }
    lastCheck = millis();
  }
}
```

These are the error message when i upload to Arduino Mega 2560 R3:
```
C:\Users\user\Documents\Arduino\Teknofest\WorkingGPS\WorkingGPS.ino: In function 'void setup()':
C:\Users\user\Documents\Arduino\Teknofest\WorkingGPS\WorkingGPS.ino:16:45: error: no matching function for call to 'HardwareSerial::begin(int, int, int, int)'
   Serial2.begin(9600, SERIAL_8N1, RXD2, TXD2);
                                             ^
In file included from C:\Users\user\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6\cores\arduino/Arduino.h:233:0,
                 from C:\Users\user\AppData\Local\arduino\sketches\35F008C038F4D5F98D48C6B747722CC2\sketch\WorkingGPS.ino.cpp:1:
C:\Users\user\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6\cores\arduino/HardwareSerial.h:121:10: note: candidate: void HardwareSerial::begin(long unsigned int)
     void begin(unsigned long baud) { begin(baud, SERIAL_8N1); }
          ^~~~~
C:\Users\user\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6\cores\arduino/HardwareSerial.h:121:10: note:   candidate expects 1 argument, 4 provided
C:\Users\user\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6\cores\arduino/HardwareSerial.h:122:10: note: candidate: void HardwareSerial::begin(long unsigned int, uint8_t)
     void begin(unsigned long, uint8_t);
          ^~~~~
C:\Users\user\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6\cores\arduino/HardwareSerial.h:122:10: note:   candidate expects 2 arguments, 4 provided
exit status 1

Compilation error: no matching function for call to 'HardwareSerial::begin(int, int, int, int)'
```